MarlinFirmware/Marlin #6456: How to allow dual Z on RAMPS 1.4?

Nicolayka, 2017-04-26:
Help please!

// A single Z stepper driver is usually used to drive 2 stepper motors.
// Uncomment this option to use a separate stepper driver for each Z axis motor.
// The next unused E driver will be assigned to the second Z stepper.
#define Z_DUAL_STEPPER_DRIVERS

#ifdef Z_DUAL_STEPPER_DRIVERS
  #undef EXTRUDERS
  #define EXTRUDERS 1
#endif

thinkyhead, 2017-04-26:
You've enabled the option, and are doing great so far.

What is your specific question?

Nicolayka, 2017-04-26:
Tried to enable the "#define Z_DUAL_STEPPER_DRIVERS", from which the second motor is operating, but he has mad speed.

thinkyhead, 2017-04-26:
Has he got the same micro-stepping jumpers installed under both stepper drivers?

linux01d, 2017-05-11:
I have MKS Base v1.5 with 5drivers on my Sunhokey Prusa i4 (clone of prusa i3).
I have the same problem with Marlin 1.1.0, one motor turns much more faster than «original» z-axis, being connected to «Z-mot » port.
The same hw works very well with Marlin 1.0.0, I've changed firmware several times and ensured that it's software problem, no stepping motor adjustment needed.

Bob-the-Kuhn, 2017-05-11:
I just tried Marlin-bugfix-1.1.x from yesterday and my dual Z is working correctly.

Did you transfer your machine specific items to the config files that came with 1.1.0 ?  If you just dropped in the ones from RC8 and older can get strange results (and a lot of compiler errors).

If your config files are up to date then please post them here.  Just ZIP them up and drop them on your reply.

linux01d, 2017-05-12:
Yes, [I cooked it from the scratch](https://github.com/linux01d/Marlin/commit/66376ac5ff3586304773416df7dbb85862ce7ca8) :) One-by-one, taking care about deprecated parameters.
Also, I can upload ZIP, if it still needed :)
Thanks!

linux01d, 2017-05-15:
I just tried Marlin bugfix-1.1.x (c262ea92e00b9bc8e8935f46e4dc6fd0e4066486) with [this configuration](https://github.com/linux01d/Marlin/commit/9df4153ccff3a9bf8bc29f0fac3f69b24666c966) with the same result :(

Bob-the-Kuhn, 2017-05-15:
**Z_DUAL_STEPPER_DRIVERS** needs to be enabled in configuration_adv.h

linux01d, 2017-05-16:
> Z_DUAL_STEPPER_DRIVERS needs to be enabled in configuration_adv.h
[Here it is](https://github.com/linux01d/Marlin/commit/9df4153ccff3a9bf8bc29f0fac3f69b24666c966#diff-18dab9d337cee2f9d2209940ed08dbacR266), I suppose.

Bob-the-Kuhn, 2017-05-16:
Correct

linux01d, 2017-05-16:
Should I open different issue for my case?

Bob-the-Kuhn, 2017-05-16:
No need to open another issue.

Is your dual Z working properly?

linux01d, 2017-05-16:
> Is your dual Z working properly?

Nope.

Bob-the-Kuhn, 2017-05-16:
Time for some sanity checks.

Please send a photo of how the Z2 motor is attached to the controller.

Disconnect the two Z motors from the belts/screws so the motors can turn freely.
- Do both motors turn in the same direction at the same speed?
- Swap the cables for the two Z motors AT THE CONTROLLER.  Do both motors turn in the same direction at the same speed?

linux01d, 2017-05-24:
> Please send a photo of how the Z2 motor is attached to the controller.
Connected to [E1-mot](https://yadi.sk/i/Vnlo9odu3JVT52) port.
> Disconnect the two Z motors from the belts/screws so the motors can turn freely.

>    Do both motors turn in the same direction at the same speed?
Same direction, but different speed.
>    Swap the cables for the two Z motors AT THE CONTROLLER. Do both motors turn in the same direction at the same speed?
Same direction, different speed.

[Video](https://yadi.sk/i/6dxJUZ0N3JVYt4)

linux01d, 2017-05-24:
[MKS Base v1.5](https://yadi.sk/i/XYeWinHV3JVZ9P)

Bob-the-Kuhn, 2017-05-25:
Marlin 1.0.? works
Marlin 1.1.? Z motors spin at different speeds in same direction
bugfix-1.1.x Z motors spin at different speeds in same direction

You've definitely got me scratching my head.  

Here's a long shot.  I've copied the RAMPS section out of the firmware from a [MKS reseller site](http://osoyoo.com/2016/06/30/mks-1-4-3d-printer-board-marlin-firmware-installation-guide/) .  See if it's better behaved with this file when using bugfix-1.1.x
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1027673/pins_RAMPS.zip)

linux01d, 2017-05-25:
I have spare board MKS Base v1.5 and can make more photos at any time (just ask), but don't have motors for experiments, I'll have to use my printer.

Bob-the-Kuhn, 2017-05-25:
Before I kick this up to more experienced people, lets see if we can better identify when things went wrong.

Please try [Marlin RC8](https://github.com/MarlinFirmware/Marlin/archive/1.1.0-RC8.zip).  Unfortunately it means you'll have to translate the config files as names and options have changed between RC8 & 1.1.x.

Bob-the-Kuhn, 2017-05-26:
I realized today that in the video that the one Z motor was running a lot faster with 1.1.x than with 1.0.x.

I can't figure out why that's happening.  I used your configuration files, downloaded it and printed out a list of the pins and the functions assigned to them.  Z_STEP and E1_STEP have no other functions assigned to those pins.

Besides trying RC8 I'd also like you to try the following with bugfix-1.1.x:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.4, 80.4, 400*1.010, 400*1.010 }
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 404, 96 }

Please also see if it's the Z or the E1 channel that's spinning too fast.  Just unplug one & see if the other is spinning at the normal or the fast speed.

Bob-the-Kuhn, 2017-05-26:
@thinkyhead , @Roxy-3D  - I'm out of ideas on this one.  A fresh perspective is needed.

He's running dual Z drivers on a MKS Base v1.5 controller and seeing the following:
- Marlin 1.0.? - both Z steppers rotate properly
- Marlin 1.1.0 - one Z stepper rotates much faster than the other, the other is rotating at the 1.0.? speed
- Marlin bugfix-1.1.x - same problem as Marlin 1.1.0
- When I issue M43 I, there are no extra functions assigned to the Z and the E1/Z2 pins.
- The speed issue follows the channel.  Swapping stepper cables moves the extra speed to the other stepper.

Since it's an MKS product we can't get a schematic for it.  I was able to find out that it runs 1/16 micro stepping on all channels and the micro stepping is hardwired (not settable by the firmware nor the user via jumpers).

The driver ICs are soldered to the board.

There's not an obvious firmware reason why they'd rotate at different speeds since both step pins are written with the same macro.

The only thing I can think of that the firmware change might have affected is the step pulse width to the controller chips.  Usually if the pulse is too narrow then we'd be loosing steps.  We could set the step pulse width to 100uS and see if that fixes it.

Maybe there's some marginal hardware.  
- If the stepper current is too low then we'd be loosing steps, not gaining them.  Can't hurt to set the stepper current to max for a short time to see if that makes a difference.
- Signal interference between cables?

Another really far out idea would be to play with the pin assignments and see if we can find a pair of channels that rotate at the same speed.

linux01d, 2017-05-26:
In the video at 2:36 you can notice how easy the motor stops by hand, with a simple touch, the torque is minimal there.

linux01d, 2017-05-26:
@Bob-the-Kuhn, I'll try your seggestions ASAP.

Bob-the-Kuhn, 2017-05-26:
A lot faster (4x-10x?) with little torque.  

I was just looking through the A4988 data sheet and it'll try to recover from an over current event every 20-40uS.  I wonder if this is why there are apparently more steps than should be.

The over current threshold is dependent on the Vref setting.  Increasing Vref might actually be a solution.  Maybe the pot has some corrosion/dirt in it.  Going back and forth between the extremes a few times is usually enough to clear the corrosion/dirt out.

You can also try setting MINIMUM_STEPPER_PULSE to 10.    1 is the minimum for that chip.  Don't know what the pulse width is when it's set to zero.

If turning the current up and setting MINIMUM_STEPPER_PULSE to 10 doesn't help then you could try moving the logical stepper channels to different sockets.  The ZIP file contains pins_RAMPS.h files with that done.  Save your current pins_RAMPS.h file and then drop in one out of the ZIP file.  All you need to do is swap the cables
[pins_RAMPS.h.swapped.zip](https://github.com/MarlinFirmware/Marlin/files/1030889/pins_RAMPS.h.swapped.zip)

linux01d, 2017-05-26:
Hmmm, I'm sory, but I didn't mentioned yet, that I used Marlin 1.0.0 provided by Sunhokey.
May be their engineers modified some settings elsewhere excepting Configuration.h and Configuration_adv.h.
[Marlin firmware 1.0.x by Sunhokey](Sunhokey).

Bob-the-Kuhn, 2017-05-26:
It's possible.  Digging it out would be a challenge.

Time to call it a night.

Talk to you tomorrow.

Bob-the-Kuhn, 2017-05-31:
Status?

Roxy-3D, 2017-05-31:
>@Roxy-3D - I'm out of ideas on this one. A fresh perspective is needed.
> He's running dual Z drivers on a MKS Base v1.5 controller and seeing the following:

I'm sorry.  I apologize.  The way I read emails and issues caused me to miss this one.  I didn't read this issue even though you flagged me on it.

The Dual Z-Motors is an example of  Marlin code where I know the functionality is there but I've never used it or looked at it...    (You do remember me saying:   Nobody can even know 1/2 the details of the Marlin code base???)    But if you want, I'll start digging in and we can bounce ideas back and forth.

Bob-the-Kuhn, 2017-05-31:
My best guess is this is a hardware issue.  Either the default stepper pulse width is marginal for the stepper IC or the current  limit needs to be set higher.

Roxy-3D, 2017-05-31:
> or the current limit needs to be set higher.

I always assume this first.  I turn the power up so I can just barely hold the stepper motor.       A lot of times, that fixes the problem.

thinkyhead, 2017-05-31:
I'm not sure where pulse width would affect torque. It should have no effect on holding torque, at any rate….

Next, try swapping the stepper drivers and see if the problem still appears on the same stepper motor.

Bob-the-Kuhn, 2017-05-31:
They're soldered to the board.

Bob-the-Kuhn, 2017-05-31:
He's swapped stepper cables at the controller.  The problem moved to the other stepper so the problem stays with the channel.

Grogyan, 2017-05-31:
It does sound like one of the stepper drivers us going into thermal
shutdown.
Suggest calibrating the Vref for both drivers so that they are equal

linux01d, 2017-05-31:
> Status?
I'm on a long-time business trip right now. I'll be back only 22.05

> Suggest calibrating the Vref for both drivers so that they are equal

They are equal. I've adjusted them with voltmeter. 0.9v on potentiometers is enough and works fine under Marlin 1.0.x.

linux01d, 2017-05-31:
> It does sound like one of the stepper drivers us going into thermal shutdown.

All drivers has heatsinks and 80mm fan over them. No thermal issues possible, I measured temperature with IR-thermometer, it just 40°C.

Bob-the-Kuhn, 2017-05-31:
@linux01d - hope you're not stuck in the hotel room every night. Gets boring really quick especially when you don't speak the language.

thinkyhead, 2017-05-31:
Did you try changing `MINIMUM_STEPPER_PULSE` and see no effect? It's meant to add extra time, but only if extra time is needed. So, apart from being precise, it depends on having a correct estimate of the number of cycles that is already eaten up by code between the start and end of a pulse. The more steppers the `Stepper::isr` has to process, the more likely it is that the proper amount of delay is being applied already. Also, if other steppers are behaving properly then chances are there's enough delay, and `MINIMUM_STEPPER_PULSE` won't have any effect.

Still, it can be revealing to mess with `MINIMUM_STEPPER_PULSE` to see whether it does have any effect. This might help rule out the length of pulses as being involved, or show at least how changing stepper pulse length might affect steppers — not just the problem one. The low range would be 9 or 10ms, and the higher range around 15. If you set it to 20 or more you'll hit a point of failure.

Anyway, the code pertaining to pulsing the Z2 stepper is indeed unchanged from the code in earlier versions of Marlin. Without knowing some obvious difference it's a tricky problem to narrow down.

Roxy-3D, 2017-05-31:
The other thing I do when a motor loses position is turn the acceleration and max feed rate way down.  That may not be the problem, but it usually helps me isolate the problem.

linux01d, 2017-06-27:
I'm still alive and I'm going to try your suggestions at weekend.
Thanks for patience!

linux01d, 2017-07-04:
MINIMUM_STEPPER_PULSE=10 with original pins_RAMPS.h doesn't help.
MINIMUM_STEPPER_PULSE=15 with original pins_RAMPS.h doesn't help.

----

MINIMUM_STEPPER_PULSE=0 with "pins_RAMPS.h.swapped E0 & E1" works, axis Z is ok. Problem moved to extruder (as expected, cause cables swapped).

----

MINIMUM_STEPPER_PULSE=0 with "pins_RAMPS.h.swapped Z & E0", axis Z is NO ok, one motor (on channel E1) turns much faster.

linux01d, 2017-07-04:
I have spare MKS Base v1.5. Is there a sense to try it?

FHeilmann, 2017-07-04:
I checked the PIN assignments of the Sunhokey fork against the latest Marlin, they are identical so we can assume that the MKS1.5 at least as far as steppers go, is wired up like a MKS1.3.

before you attempt the solution below: Can you tell which of the Z-Steppers is turning too fast? (The one connected to Z or the one connected to E1?)

If it is the stepper connected to E1, you can maybe try to check for a hardware issue by making the following changes to `pins_RAMPS.h`: 

Change:

```C
#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_CS_PIN           40

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#define E0_CS_PIN          42

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#define E1_CS_PIN          44
```

to 

```C
#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_CS_PIN           40

#define E1_STEP_PIN        26
#define E1_DIR_PIN         28
#define E1_ENABLE_PIN      24
#define E1_CS_PIN          42

#define E0_STEP_PIN        36
#define E0_DIR_PIN         34
#define E0_ENABLE_PIN      30
#define E0_CS_PIN          44
```

and plug your second Z-stepper into your extruder socket (labeled E0) and the extruder-stepper into the socket for the second extruder (labeled E1).

This change will move the second Z motor to another driver, so if everything works after this change, your E1 stepper driver is faulty. If the problem still persists, it is a software issue.

linux01d, 2017-08-07:
> #define Z_STEP_PIN         46
> #define Z_DIR_PIN          48
> #define Z_ENABLE_PIN       62
> #define Z_CS_PIN           40
>
> #define E1_STEP_PIN        26
> #define E1_DIR_PIN         28
> #define [...]

I've tried this on BOTH CONTROLLERS I have :) They works with not difference and:
1. Swapped E0 and E1 brings Z-axis work ok, but extruder turns much faster, than should to.
2. Marlin v1.0.1 hacked by chinese Sunhokey works fine without issues.

I ensured with this experiment that both controllers are ok, but E1 port had to be tuned some other way as Sunhokey do.

Any ideas here?

linux01d, 2017-10-14:
ping :)

linux01d, 2017-10-14:
I think that MKS Base v1.5 needs special customization for E1 in case of Z2 usage.

linux01d, 2017-10-25:
:(

disco22522, 2017-10-28:
I'm trying to use the dual z motor function, where e1 is the other stepper driver. I have RAMPS 1.4 with mega2560. Both of the motors work, but only if one or the other is plugged into the z-axis pins. Uncommenting #define Z_DUAL_STEPPER_DRIVERS does not seem to work, and defining the pins in pins.h with "36, 34, 30" for step, dir, enable, does not work.

fiveangle, 2017-10-28:
> 1. Swapped E0 and E1 brings Z-axis work ok, but extruder turns much faster, than should to.
> 2. Marlin v1.0.1 hacked by chinese Sunhokey works fine without issues.

Schematic of MKS Base shows it has MCU-addressable microsteping for the A4988 but I don't see any config for this in the pins file.  Looks like someone needs to fix the MKS base pins file to support it.
[image]
[PAGE1 (3 files merged).pdf](https://github.com/MarlinFirmware/Marlin/files/1424415/PAGE1.3.files.merged.pdf)

Where is the source for the original Marlin that was on it ?

linux01d, 2017-10-29:
> Where is the source for the original Marlin that was on it you said was working ?
[This comment.](https://github.com/MarlinFirmware/Marlin/issues/6456#issuecomment-304205561)

fiveangle, 2017-10-29:
Okay, I only looked at it for a second earlier and saw MS1/MS2 on the sch. but on closer inspection, it appears they've earmarked 5/15, 92/93 for X_MS1/2, Y_MS1/2 only (not Z, E0, E1) but as they have no config for those pins in the release, I suspect the microstepping was scrapped entirely and all A4982 are hard-wired for 16 micro steps by pulling both MS1/2 high:
[image]
 so that was just a wild goose chase.

Are you 100% confident you defaulted the EEPROM contents after upgrade ?  Marlin should detect if it's not the write layout and default it, but not everything always goes to plan.

If that doesn't resolve I'm having a hard time beleiving this works on the old Marlin (I couldn't see any of the videos or earlier files as they all come up deleted).

Regardless, if swapping E0 and E1 makes your Z work, then it sounds like you're golden and you just need to re-calibrate your extruder steps/mm setting so it's not spining so fast.

-=dave

disco22522, 2017-10-29:
You have to go into Marlin with Arduino, go to pins_ramps.h and type out the pins for "Z2" as E1 pins. This wasn't readily apparent to me even in some guides so I hope this helps someone getting theirs to work.

NiftyMaker, 2017-12-17:
Hey @linux01d, did you solved it? I'm having the same issue...

mrJefferson105, 2017-12-23:
I had the same problem with marlin 1.1x but  all I had to do was to replace the driver as the current was not set properly and it worked . also on the configuration.h I defined to use dual z steppers .
#define Z_DUAL_STEPPER_DRIVERS
I only used this line no extra stuff.

thinkyhead, 2017-12-25:
> You have to go into Marlin with Arduino, go to pins_ramps.h and type out the pins for "Z2" as E1 pins. 

For future reference, people, you don't. Just set `Z_DUAL_STEPPER_DRIVERS` and your next set of unused E pins will be auto-assigned to your Z2 pins. Done.

phantom-code, 2017-12-28:
I tried to flash the latest Marlin 1.1.8 to my Sunhokey Prusa i4 and encountered the same issue. I spent 2 days analyzing what's going on and found that the ms2 pin on E1 stepper's driver is pulled down:

 [image]

Thus, according to the A4982 datasheet, it works in a "Half Step" mode.
[image]
It seems that it is initialized wrong somewhere. The codebase is tremendously hard to read because of the insane preprocessor usage, so I couldn't find the exact place where it happens, but I think I was able to make a rough workaround. I just switched the appropriate pin to the output mode and set it to the high state. The patch is here, but please, try it at your own risk. I haven't tested it enough.
```--- ./stepper.original.cpp	Mon Dec 25 07:52:04 2017
+++ ./stepper.cpp	Thu Dec 28 20:56:28 2017
@@ -1102,14 +1102,18 @@
   #endif
   #if HAS_E3_STEP
     E_AXIS_INIT(3);
   #endif
   #if HAS_E4_STEP
     E_AXIS_INIT(4);
   #endif
+  
+	SET_OUTPUT(4);
+	OUT_WRITE(4, HIGH);
+
 
   // waveform generation = 0100 = CTC
   SET_WGM(1, CTC_OCRnA);
 
   // output mode = 00 (disconnected)
   SET_COMA(1, NORMAL);
 

```

fiveangle, 2017-12-28:
@phantom-code - pins for all boards are contained in pins.h:

```
#elif MB(MKS_BASE)
  #include "pins_MKS_BASE.h"          // ATmega1280, ATmega2560
```

`pins_MKS_BASE.h` has only these changes from `pins_RAMPS.h`:

```
#if HOTENDS > 2 || E_STEPPERS > 2
  #error "MKS BASE 1.0 supports up to 2 hotends / E-steppers. Comment out this line to continue."
#endif

#define BOARD_NAME "MKS BASE 1.0"

//
// Heaters / Fans
//
// Power outputs EFBF or EFBE
#define MOSFET_D_PIN 7

#define CASE_LIGHT_PIN            2

//
// M3/M4/M5 - Spindle/Laser Control
//
#define SPINDLE_LASER_PWM_PIN     2  // MUST BE HARDWARE PWM
#define SPINDLE_LASER_ENABLE_PIN 15  // Pin should have a pullup!
#define SPINDLE_DIR_PIN          19

#include "pins_RAMPS.h"
```

RAMPS has no support for MS1/MS2 driver step config:

```
//
// Steppers
//
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#define X_CS_PIN           53

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#define Y_CS_PIN           49

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_CS_PIN           40

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#define E0_CS_PIN          42

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#define E1_CS_PIN          44
```

As I mentioned in [my post above (https://github.com/MarlinFirmware/Marlin/issues/6456#issuecomment-340251855), I found no code in the MKS marlin fork that detailed the MS1/MS2 pin connections for the extruder drivers (nor for X & Y, which do appear in the sch) so I suspected they had both tied high (as I found on an MKS sch diagram above).

If you found MS1 and/or MS2 for the extruder drivers configurable via pins on the MCU, then all that should be required is to determine which connect to each, and add the logic in `pins_MKS_BASE.h` above.  This would be a big plus for many users who are struggling to attain required speed at 16/th microstepping.  [PINS_DEBUGGING feature](http://marlinfw.org/docs/gcode/M043.html) may be useful here.

-=dave

phantom-code, 2017-12-28:
@fiveangle yes, I realized the same thing: there are no MS1/MS2 pin definitions in pins_RAMPS.h nor in pins_MKS13.h. And during my investigation I found that they are tied high and at the same time they are connected to the MCU. So, if the MCU doesn't touch them, we get the 1/16 step mode. I just took the PG5 (OC0B) pin (it's connected to the MS2 input of the E1 stepper driver) and switched it to get the right mode. I put my patch here in hope it can be useful for people who have this issue. Maybe someone familiar with Marlin codebase will be able to find the reason that causes the issue or even add the microstepping support for MKS boards.

fiveangle, 2017-12-28:
If you find the pins, it would be easy to add the config

-=dave

phantom-code, 2017-12-28:
So, I took the schematic from [this comment](https://github.com/MarlinFirmware/Marlin/issues/6456#issuecomment-340204042) and correlated GPIO pin names with DIO#_PIN definitions from the fastio1280.h file
[image]
Judging by this information we should have the following:
```
#define X_MS1_PIN 5
#define X_MS2_PIN 6

#define Y_MS1_PIN 59
#define Y_MS2_PIN 58

#define Z_MS1_PIN 22
#define Z_MS2_PIN 39

#define E0_MS1_PIN 63
#define E0_MS2_PIN 64

#define E1_MS1_PIN 57
#define E1_MS2_PIN 4
```
I'm going to test it tomorrow and report the result.

[39 more replies]

closed 2018-02-14 by thinkyhead