MarlinFirmware/Marlin #6461: PT-BR language group

CoderSquirrel, 2017-04-26:
I was reading some of the pt-br translations and some of them are... understable but others are strange or not friendly, i mean, some are just strict translation of the work or "aportuguesamento" of the word which is no too ok(? i think), so, could we make a group for translation or to discuss, by this i mean just chat no discuss like figth and stuff, to make the portuguese translations a little... cooler, or even better?
what u think @esenapaj @jbrazio @hydrocat @ivan-galvez ?

hydrocat, 2017-04-27:
Sure ! I was assigned as a maintainer for the pt-br translations but I haven't been as active as I expected..

hydrocat, 2017-04-27:
By which means should we contact ? slack ?

CoderSquirrel, 2017-04-27:
I beber uses Slack but for me it's ok

CoderSquirrel, 2017-05-01:
never *

hydrocat, 2017-05-02:
Me neither but it's what everyone is using now. What do you suggest ?

CoderSquirrel, 2017-05-04:
google groups maybe? 
but no probleming in using stack, is always good to learn new stuff

loos-br, 2017-05-05:
is this still an issue ?

closed 2017-11-01 by thinkyhead