MarlinFirmware/Marlin #6464: need several improvements

Cyper77, 2017-04-26:
first start on delta kossel in the LCD move axis menu;
-if you want to move head(Z) to top, LIMIT SWITCHes does not work,
-if you want to feed PLA at first time or test motors, menu needs to be improve, because on prepare menu-move axis-extruder does not work without heating.  move extruder function must be bonded with heat function.
-on my anycubic delta, some slicers auto generated gcode and it have heat bed commands, if you dont connect your thermistor than send to printer this code, marlin fw stuck and comes fatal error.
-also DELTA_HOME_TO_SAFE_ZONE parameter absent in calibration header. (if this useful)
thank you for great firmware for delta's.

thinkyhead, 2017-04-26:
1. Use `G28` or enable `MAX_SOFTWARE_ENDSTOPS`. The limit switches are not designed to prevent movement, as they are not enabled during normal operation, only during `G28` or when printing from SD with the option to abort on endstop hit.
2. This is on our to-do list already.
3. That was already fixed. Apparently you are using RC8 or older.
4. `DELTA_HOME_TO_SAFE_ZONE` is a configuration option. What is the "calibration header"?

Cyper77, 2017-04-27:
thank you, yes i am using rc8 and sorry this is my mistake, file is configuration.h for calibration :)
 DELTA_HOME_TO_SAFE_ZONE not adjustable in this file ;)

closed 2017-11-01 by thinkyhead