MarlinFirmware/Marlin #6466: Usage of Z Probe together with the Z Min Endstop

Rigid3D, 2017-04-27:
On the current code when Z_MIN_PROBE_ENDSTOP is used. During G29 operation triggering either Z Min Endstop or Z Probe switch acts as detection of the bed.

When both Z Min and Z Probe is both available on the machine, I expect the Z Min Endstop to be used during G28 and Z probe to be used on G29 operation.

Since I did not know whether this is done intentionally like this, I did not create a pull request. I have traced the code and just a small change is needed in endstops.cpp, if we can agree on the logic I can create a pull request.

My case was as follows:

The machine uses metal bed and the contact between the tip of the nozzle and the bed can be used as a Z probe when the nozzle and the bed is clean. Without filament and with clean bed ABL can be run and the leveling matrix can be kept in the eeprom. On the start of each print homing can be handled by the reliable Z Min Endstop and from time to time ABL can be run with clean nozzle and bed.

With the current logic if Z Min Endstop (optoswitch) is triggered before Z Probe switch, it is inferred as bed is probed and a false ABL matrix is created.

Bob-the-Kuhn, 2017-04-27:
The implication is that the Z probe function isn't reliable unless you do some preparation.  Correct?

Tannoo, 2017-04-27:
// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS

Rigid3D, 2017-04-27:
@Bob-the-Kuhn Exactly. To compensate for the not planar bed non-linear ABL will be used. G29 will be done from time to time after cleaning the bed and the nozzle. G28 will be handled by Z min endstop G29 will be handled by the Z probe.

@Tannoo That's not the problem. The problem is in the logic of endstops.cpp

Starting at line 394 for of the endstops.cpp the code is like this:

      #if HAS_Z_MIN
        #if ENABLED(Z_DUAL_ENDSTOPS)

          UPDATE_ENDSTOP_BIT(Z, MIN);
          #if HAS_Z2_MIN
            UPDATE_ENDSTOP_BIT(Z2, MIN);
          #else
            COPY_BIT(current_endstop_bits, Z_MIN, Z2_MIN);
          #endif

          test_dual_z_endstops(Z_MIN, Z2_MIN);

        #else // !Z_DUAL_ENDSTOPS

          #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
            if (z_probe_enabled) UPDATE_ENDSTOP(Z, MIN);
          #else
            UPDATE_ENDSTOP(Z, MIN);
          #endif

        #endif // !Z_DUAL_ENDSTOPS

      #endif // HAS_Z_MIN

      // When closing the gap check the enabled probe
      #if ENABLED(Z_MIN_PROBE_ENDSTOP)
        if (z_probe_enabled) {
          UPDATE_ENDSTOP(Z, MIN_PROBE);
          if (TEST_ENDSTOP(Z_MIN_PROBE)) SBI(endstop_hit_bits, Z_MIN_PROBE);
        }
      #endif

If thiss part is updated like below, the work logic will be like my proposal
i.e. G28 ->Z Min Endstop G29->Z Probe
The lines with SUTAS comments are added.

     #if HAS_Z_MIN
        #if ENABLED(Z_DUAL_ENDSTOPS)

          UPDATE_ENDSTOP_BIT(Z, MIN);
          #if HAS_Z2_MIN
            UPDATE_ENDSTOP_BIT(Z2, MIN);
          #else
            COPY_BIT(current_endstop_bits, Z_MIN, Z2_MIN);
          #endif

          test_dual_z_endstops(Z_MIN, Z2_MIN);

        #else // !Z_DUAL_ENDSTOPS

          #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
            if (z_probe_enabled) UPDATE_ENDSTOP(Z, MIN);
          #elif ENABLED(Z_MIN_PROBE_ENDSTOP)                              //SUTAS
            if (z_probe_enabled) {                                               //SUTAS
              UPDATE_ENDSTOP(Z, MIN_PROBE);                                             //SUTAS
              if (TEST_ENDSTOP(Z_MIN_PROBE)) SBI(endstop_hit_bits, Z_MIN_PROBE);               //SUTAS
            } else {
                  UPDATE_ENDSTOP(Z, MIN);
                  }                                                                                         //SUTAS
          #else
            UPDATE_ENDSTOP(Z, MIN);
          #endif   //Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

        #endif // !Z_DUAL_ENDSTOPS

      #endif // HAS_Z_MIN

      // When closing the gap check the enabled probe
      #if ENABLED(Z_MIN_PROBE_ENDSTOP)                           
        if (z_probe_enabled) {
          UPDATE_ENDSTOP(Z, MIN_PROBE);
          if (TEST_ENDSTOP(Z_MIN_PROBE)) SBI(endstop_hit_bits, Z_MIN_PROBE);
        }
      #endif

Bob-the-Kuhn, 2017-04-27:
I know we've had at least one person trying to use this feature to act as a backup in case the Z probe failed.

The options I see when both Z_MIN_PIN and Z_MIN_PROBE_ENDSTOP are enabled:
- Check both Z_MIN_PIN and Z_MIN_PROBE_PIN when probing.  This is the current system.
- Check only Z_MIN_PROBE_PIN when probing.  This is the request.
- Make this (yet another) configuration item.

Tannoo, 2017-04-28:
So, are we saying that this is incorrect?
```
 *   All Z PROBE pin options are configured by defining (or not defining)
 *   the following five items:
 *       Z_MIN_PROBE_ENDSTOP – defined below
 *       Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN – defined below
 *       Z_MIN_PIN - defined in the pins_YOUR_BOARD.h file
 *       Z_MIN_PROBE_PIN - defined in the pins_YOUR_BOARD.h file
 *
 *   If you're using a probe then you need to tell Marlin which pin to use as
 *   the Z MIN ENDSTOP.  Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN determines if the
 *   Z_MIN_PIN or if the Z_MIN_PROBE_PIN is used.
 *
 *   The pin selected for the probe is ONLY checked during probing operations.
 *   If you want to use the Z_MIN_PIN as an endstop AND you want to have a Z PROBE
 *   then you’ll need to use the Z_MIN_PROBE_PIN option.
 *
 *   Z_MIN_PROBE_ENDSTOP also needs to be enabled if you want to use Z_MIN_PROBE_PIN.
 *
 *   The settings needed to use the Z_MIN_PROBE_PIN are:
 *       1. select the type of probe you're using
 *       2. define Z_MIN_PROBE_PIN in your pins_YOUR_BOARD.h file
 *       3. disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *       4. enable Z_MIN_PROBE_ENDSTOP
 *   NOTE – if Z_MIN_PIN is defined then it’ll be checked during all moves in the
 *          negative Z direction.
 *
 *   The settings needed to use the Z_MIN_PIN are:
 *       1. select the type of probe you're using
 *       2. enable Z_MIN _PIN in your pins_YOUR_BOARD.h file
 *       3. enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *       4. disable Z_MIN_PROBE_ENDSTOP
 *   NOTES – if Z_MIN_PROBE_PIN is defined in the pins_YOUR_BOARD.h file then it’ll be
 *          ignored by Marlin
 */
```

Bob-the-Kuhn, 2017-04-28:
It's true.  If Z_MIN_PROBE_ENDSTOP is enabled then both the probe and the Z_MIN_PIN will be checked.

tcm0116, 2017-05-04:
> I know we've had at least one person trying to use this feature to act as a backup in case the Z probe failed.

Me!

> On the current code when Z_MIN_PROBE_ENDSTOP is used. During G29 operation triggering either Z Min Endstop or Z Probe switch acts as detection of the bed.

In my case, that's exactly what I wanted. If the Z Probe failed for some reason, I wanted the Z Min switch to stop the nozzle from pushing too far into the bed. In order to keep the Z Min switch from interfering with the Z Probe, I set the Z Min switch such that it would be triggered with the nozzle about 1mm lower than the surface of the bed. In my case, I was able to home the Z axis with the nozzle outside of the bed area. 

@Rigid3D would it be possible to try lowering your Z Min switch? You'll have to ensure that the Z axis can be leveled with the nozzle outside of the bed area.

Rigid3D, 2017-05-04:
@tcm0116 This logic is not a must for me. I was just experimenting some different abl setups and seen the current logic. For now the logic that I have proposed is not needed. If I need it in the future I will create a pull request with  a switch in the configuration.h that enables the user to choose the logic that best fits needs.

closed 2017-05-04 by Rigid3D

kulfuerst, 2019-04-04:
I'm currently encountering the issue that my z_min endstops interfere with probing just as described above. How was this issue solved here?