MarlinFirmware/Marlin #6467: Possible GPL voliation - Monoprice MP Select Mini

mmonsterski, 2017-04-27:
Source directory: [Motion firmware](https://drive.google.com/drive/u/0/folders/0BxyFI3iDaicLQXRhNi1IZkd4WU0)
Filename: motion_29.zip

In my opinion this version is based on heavy modified version with "V23" version of configuration_store:
>V23 V23 Settings Stored (    bytes) V23  stored settings retrieved (    Hardcoded Default Settings Loaded

Roxy-3D, 2017-04-28:
Yeah....  But can you be more specific?   They are publishing the source.  Isn't it OK if they leverage our work to get to where they want to be?

mmonsterski, 2017-04-28:
Could You provide me address to download source because I'be looking yesterday few hours and I don't found it?

closed 2017-11-01 by thinkyhead

Saturn49, 2020-04-25:
Bumping this - I do not believe Monoprice (nor the manufacturer, Malyan) publishes the source for the Monoprice Mini Select (commonly known as the MPMS) or what Malyan calls M200.

They have several variations, here's one:
https://www.monoprice.com/product?p_id=15365

They have linked firmware (binaries) but no source that I've been able to find.