MarlinFirmware/Marlin #6468: MSG_THERMAL_RUNAWAY during printing, job aborted

khschmidt, 2017-04-27:
What is the reason for this error message ( I suspect, that the nozzle temperatur is out of range)
it was near the end of a about 4 hour print. It was the first layer after 30% infill with honey comb and cooling was activated. Maybe the nozzle Temperatur was cooling down due the cooling fan and the too slow PID Control.

My question: Is there a Parameter for the range of allowed temperatur deviation?
Is it possible, not to switch off in the software but to wait until the temperatur is again within the allowed range?

Tannoo, 2017-04-27:
> Is there a Parameter for the range of allowed temperatur'e' deviation?

Yes
```
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius
```
>wait until the temperatur is again within the allowed range?

Only during initial heating with `M109 Sxxx`.
Not with `THERMAL_PROTECTION_HOTENDS`. Safety feature.

khschmidt, 2017-04-27:
Thank you,

that's the Information I need :-)

I reduce the bridges fan speed in slicer down to 60% and increase the thermal_protection_hysteresis to 7 degrees.

psavva, 2017-05-01:
@khschmidt can you close this ticket if all is good :)

closed 2017-05-02 by khschmidt