MarlinFirmware/Marlin #6469: X endstop not showing following M119 firmware upgrade to 1.1.0 RC8

Part0, 2017-04-27:
Hi,
I have a ATmega 1284 based Gen 7 AVR 1.5 main board which has been running on Marlin 1.0.2 firmware without issues for a few years. I tried to updated to Marlin 1.1.0 RC8. using Arduino IDE 1.8.1 All went fine with the update and everything seems to work except the X endstop was not recognised with a M119. See below

SENT: M119
READ: Reporting endstop status
READ: y_min: TRIGGERED
READ: z_min: open
READ: ok

I know the problem is firmware related because when I went back to ver 1.0.2 the x endstop is reported as expected following a M119. I tried upgrading to ver. 1.1.0-RC7 and the problem is solved. Reloading RC8, the issue is back again. Stepping back to RC7 and all is well again.

Tannoo, 2017-04-27:
Check these settings:
```
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
```

Part0, 2017-04-28:
Check these settings:

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

Thanks Tannoo for the response but there is no difference in the above settings between V1.1.0- RC7 and RC8 as tested on my printer. Also all my endstops are the same and any configuration of X-endstop in my firmware is always the same as Y and Z endstop so why is Y and Z showing up and X not in response to M119 with RC7 and not with RC8??

Bob-the-Kuhn, 2017-04-28:
I was able to reproduce this issue.

The problem only occurs if the pin number is set to zero.

We're getting ready to release Marlin 1.1.0 in a few days.  With any luck I can get this fix installed before then.

----

This is a pre-processor issue.  The HAS_xyz_MIN & HAS_xyz_MAX macros aren't fully evaluated when deciding if  `#if HAS_X_MIN` is true or not. The problem is that the pre-processor hasn't yet evaluated the Z2_MIN_PIN and Z2_MAX_PIN macros at this point so they temporarily exist with a value of zero.

In Conditionals_post.h

Replace:
`  #define IS_Z2_OR_PROBE(P) (P == Z2_MIN_PIN || P == Z2_MAX_PIN || P == Z_MIN_PROBE_PIN)`

With:
```
  #define IS_Z2_OR_PROBE(P) (PIN_EXISTS(Z2_MIN_PIN) && (P == Z2_MIN_PIN) \
                          || PIN_EXISTS(Z2_MAX_PIN) && (P == Z2_MAX_PIN) \ 
                          || PIN_EXISTS(Z_MIN_PROBE_PIN) && (P == Z_MIN_PROBE_PIN))
```

Bob-the-Kuhn, 2017-04-28:
Pull request #6490 implements the correction.

closed 2017-04-30 by thinkyhead