MarlinFirmware/Marlin #6472: Compiler warning

JimBrown, 2017-04-27:
I've been seeing this warning in the latest BugFix (I update it almost every day) for a few weeks now. Compiling with Arduino IDE 1.8.2.

```
In file included from C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\MarlinConfig.h:27:0,

                 from C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\Marlin.h:36,

                 from C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\stepper.cpp:47:

C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\stepper.cpp: In static member function 'static void Stepper::init()':

C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\macros.h:123:37: warning: invalid conversion from 'volatile void*' to 'void*' [-fpermissive]

 #define ZERO(a) memset(a,0,sizeof(a))

                                     ^

C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\stepper.cpp:1102:5: note: in expansion of macro 'ZERO'

     ZERO(e_steps);

     ^

In file included from C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\Marlin.h:28:0,

                 from C:\Users\jbwin\AppData\Local\Temp\arduino_build_210250\sketch\stepper.cpp:47:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\string.h:235:14: note: initializing argument 1 of 'void* memset(void*, int, size_t)'

 extern void *memset(void *, int, size_t);

              ^
```

Everything appears to be working ok, but I hate warnings. ;-)

Thanks.

Roxy-3D, 2017-04-27:
Agreed!   The reason the compiler is warning you about something is because it doesn't always work!!!

JimBrown, 2017-04-28:
I appreciate the support. ;-) I'm hoping someone can figure it out and fix it, because I don't know this language.

Tannoo, 2017-04-29:
I don't have that issue. Mine compiles cleanly.

autonumous, 2017-04-29:
i’ve started to have  this issue (actually as an error) when compiling on my Raspberry Pi.

autonumous, 2017-04-29:
@JimBrown @Roxy-3D 
if i add “--fpermissive” to my makefile CXTRA parameter it compiles.
(although i see that as an unclean hack) 

If i disable LIN_ADVANCE, it compiles with no errors.  
leading me to think its related to this section in stepper.cpp
```
#if ENABLED(ADVANCE) || ENABLED(LIN_ADVANCE)
    ZERO(e_steps);
    #if ENABLED(LIN_ADVANCE)
      ZERO(current_adv_steps);
    #endif
  #endif // ADVANCE || LIN_ADVANCE
```
in my last version i compiled (last month) this didn’t use the ZERO macro, but a for loop.

JimBrown, 2017-04-29:
Just found the same thing. If I disable LIN_ADVANCE, I no longer get the warning. Hopefully one of the programmers can step in and fix this.

Thanks.

autonumous, 2017-04-29:
this change was introduced by this commit https://github.com/MarlinFirmware/Marlin/commit/6a9b008044b363fb0f55aa01016afbd3347b3e68
On my system, i can confirm if revert this commit, it compiles with no warnings/errors   else i get the compile time errors 

Im guessing its something relating to having LIN_Advance enabled which  declares e_steps  as “static volatile int”, and its causing some issue with the version and configuration of the compiler i have on my Pi.   memset not liking things as volatile or something?

 if i take the same code and compile in Arduino Application on my Mac, it compiles with no issues

For now i’ll simply add the parameter (“--fpermissive”) to my MakeFile :-)

Tannoo, 2017-04-29:
>Hopefully one of the programmers can step in and fix this.

I'm sure they will, they are a little busy atm getting the next release out the door.

thinkyhead, 2017-05-01:
The warning for `volatile *` is reasonable. If the pointer might be changed in an interrupt, then it should first be copied to a non-volatile pointer, and then used in `ZERO()`. Or, if the buffer it points to might go away, then the `ZERO()` (`memset`) call should be bracketed in `CRITICAL_SECTION_START` and `CRITICAL_SECTION_END`.

PR #6515 (2017-05-01): Some minor cleanups to code formatting
- Combine grouped case statements
- Use new serial spacing function to save PROGMEM
- Adjust spacing for casts and others
- Resolve #6472
closed 2017-05-01 by thinkyhead