MarlinFirmware/Marlin #6473: 3D printer Probe offset problems

amfgames, 2017-04-27:
Hello.

I just got my probe installed on my CTC Prusa i3 using Marlin. But every time i upload my code I get the following error code:

"#error "The given BACK_PROBE_BED_POSITION can't be reached by the Z probe."

I have tried different combinations of settings without success.

My current settings:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER

 -22  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -47,6  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define ABL_GRID_POINTS_X 3
  #define ABL_GRID_POINTS_Y ABL_GRID_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 25
  #define RIGHT_PROBE_BED_POSITION 185
  #define FRONT_PROBE_BED_POSITION 58
  #define BACK_PROBE_BED_POSITION 185

#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 185
#define Y_MAX_POS 185
#define Z_MAX_POS 160
```

This is the physical postition of the hotend & inductive probe/sensor. Is this correct?
The nozzle and probe is exactly 1mm from the heatbed.
[http://bildr.no/image/WnZhMG9a.jpeg](url)

Full source code:
https://gist.github.com/anonymous/4d33bc40c36d762a251ade5810a35240

Tannoo, 2017-04-28:
`#define Y_PROBE_OFFSET_FROM_EXTRUDER -47,6` <-- Distance of probe from nozzle.
and
`#define BACK_PROBE_BED_POSITION 185` <-- Max limit for Probe.
do not jive with
`#define Y_MAX_POS 185`  <-- Max limit for Nozzle.

Your nozzle cannot go past `Y_MAX_POS` of 185. Your `Y_PROBE_OFFSET_FROM_EXTRUDER` is -47.6 in front of the nozzle. That means the nozzle needs to move another 47.6mm to get the probe there and it cannot.
So, `BACK_PROBE_BED_POSITION` should be set to no greater than 137.4. This would be the probe position, not the nozzle position. When the probe gets to this point, the nozzle will be at 185.

Tannoo, 2017-04-28:
Same goes for `X_MAX_POS` and `RIGHT_PROBE_BED_POSITION`. (If your origin is the front left of the bed.)

closed 2017-11-01 by thinkyhead