MarlinFirmware/Marlin #6474: MIxing factors

Denarian, 2017-04-27:
Can you add option to display current mixing factors on LCD ?(between hotend temp and bed temp)

boelle, 2019-06-22:
@Denarian can you explain what you mean by temp mixing factors?

InsanityAutomation, 2019-06-22:
@thinkyhead was any display adjustment made when you added gradient mode and such?

@boelle he means between the hotend and bed icons and temperature readings.

Denarian, 2019-06-22:
@boelle when using mixing hotend such as diamond hotend the extruders are moving with different speeds. Mixing factor is how much specific extruders are going to move when you for eg. want to extrude 1mm of filament. It is configured  by  M-codes M164 and M163

Vertabreak, 2020-01-19:
my A10M does this in marlin 2.0 
@boelle think this might be safe to close.

closed 2020-01-24 by boelle