MarlinFirmware/Marlin #6477: Having issue with my prints after updating to latest marlin rc8 fix.

Rhonal89, 2017-04-28:
I have a video that explain it well. First layer prints nicely on the second layer random retraction happens. https://youtu.be/FgjsdAIDIw8

Rhonal89, 2017-04-28:
Check around 2:30 in video

Roxy-3D, 2017-04-28:
That video was very helpful...   I'm not saying there isn't a problem!!!   But I've seen that too and that happens when I'm pushing the feedrate to the maximum.   I don't know for sure...  But in my case when I've seen that it is like the extruder's stepper motor isn't strong enough to push the filament at the rate specified.  And if it gets too far out of position, it is easier for the stepper motor to go the wrong way and actually go backwards.

I don't know that is what is happening.  And having that video is very helpful!   I'll work with you to figure this out and get it fixed.   But can you do a couple of things?   First, can you turn up the power to your extruder motor and see if that helps?   And if it doesn't help, can you set your Max Extruder feed rate lower...  maybe at 2/3 and see if it still happens.   

What ever extra information we can get will help us figure this out....

Rhonal89, 2017-04-28:
Sure thing. I do have a rambo board and I have the extruder set at 85 from 135 set for rambo. This is the one I have Short Body Nema 17 Bipolar Step Motor 3.5V 1A 18.4oz.in/13Ncm DIY CNC https://www.amazon.com/dp/B00PNEQ79Q/ref=cm_sw_r_cp_apa_DzSazb9T35E9P

Roxy-3D, 2017-04-28:
And up in the post up above...   I edited it to suggest seeing if a 10 C. increase in temperature helps.   Any information about what helps and what does not help will be useful to guide us!

Rhonal89, 2017-04-28:
Hi roxy, how would I go from saving my current mesh to loading back to marlin after firmware changes. Thank You!!

Roxy-3D, 2017-04-28:
Normally...  Firmware updates should not affect your saved mesh's.   But right now...  Things are in a state of flux and where everything is stored in the EEPROM keeps changing.   You can make an 'Off the Printer' backup of the mesh by saying to save it to Slot -1.   You would do a G29 S -1   and it will print the current mesh in a form you can later load it into PronterFace or Repetier Host and 'Print It'.    

Later...  You won't have to do this...  But right now you should do this and save a copy somewhere so you can easily get it back.

Rhonal89, 2017-04-29:
Hi roxy, So I up my motor power and PID tune my hotend again and up my temperature also try different file and it keep happening. So went back to another firmware rc8, I think the stable one and it's printing great.

Rhonal89, 2017-04-29:
The only reason I want to try the other firmware so bad is just because UBL. I'll see what I figure out

thinkyhead, 2017-04-29:
Sorry, the first thing we should ask is for you to test `RCBugFix` and see if we already fixed the issue. This can save us much time in discussing and investigating water under the bridge.

Roxy-3D, 2017-04-29:
> The only reason I want to try the other firmware so bad is just because UBL. I'll see what I figure out

Would it be possible to build the latest firmware with no bed leveling in it and see if you get the same problem?   It would be good to know if the problem is in the auto bed leveling code or some where else in the firmware.

Rhonal89, 2017-04-29:
Ok, so I flashed back the firmware and I let the print continue and I notice it does the retraction every second layer.

Roxy-3D, 2017-04-29:
That video was very helpful, but I could not see the extruder motor axle.   Can you stick a small piece of tape on it or something that will be visible by the camera?

Rhonal89, 2017-04-29:
Here is another video explaining a little bit more https://youtu.be/kt0XwETcLJI

Roxy-3D, 2017-04-29:
Can you change the GCode's file extension to .txt and upload it?  With a .txt extension you should be able to drag and drop it to the post.

Rhonal89, 2017-04-29:
[Raspi7inch_Case.txt](https://github.com/MarlinFirmware/Marlin/files/966309/Raspi7inch_Case.txt)

Roxy-3D, 2017-04-29:
Unfortunately...    I don't have Simplify-3D.      Can you try slicing it with Slic3r 1.2.7 or Slic3r 1.2.9 and try printing it?     It is difficult to offer support when I don't have the tools being used.

Rhonal89, 2017-04-29:
[Raspi7inch_Case.txt](https://github.com/MarlinFirmware/Marlin/files/966329/Raspi7inch_Case.txt)

Hi Roxy, here is the one with slic3r 1.29

Rhonal89, 2017-04-29:
Yes, Roxy it does the same thing as the last video. But it does something differently than simplify. It seem like it has a -z hop but is not in the slic3r setting. Ill post a video here once I restart the print.

Roxy-3D, 2017-04-29:
I might have more information...    Can you do a G26 and leave the mesh validation print on the bed.   And then do the same print where you see the funny stuff happen?    I might have duplicated what you are seeing on my printer...  If so...  It will be a lot easier to find the problem and fix it.

What I think might be happening is when the nozzle crosses a mesh line in a certain direction, the extruder is retracting...   And if it is tied to a certain direction of travel, that would explain why you are seeing it on every other layer.   The slicers alternate the direction of the fill layer by layer.

Rhonal89, 2017-04-29:
Here is a video Roxy https://www.youtube.com/watch?v=sjsBchND9L0 You can skip to 1:50 for second layer

Roxy-3D, 2017-04-29:
I only watched the first half of the video...  If there is anything important in the 2nd half please say so and I'll watch that too.     What you caught on the video is pretty close to what I saw.   Right now, I don't understand why it only happens in one direction, but when I find the bug...  It will probably make sense.  

I'm going to write an OpenScad program to generate a test pattern that should help aggravate the problem.    I would like to get this problem fixed before Thinky promotes RC-8's RCBugFix to Golden Master tomorrow.   But that may not be possible.   If the promotion happens before I find this bug...   This might be the first Bug Fix in the Golden Master!

I really really appreciate your help trying to figure this out!   Probably what is wrong is not a big deal to fix.  The problem is usually finding the bug.   But...  There is lots of debug code in place in the ubl_motion.cpp files.  So I think I'll be able to find the problem with a few hours of work...

Rhonal89, 2017-04-29:
second half shows it better

Roxy-3D, 2017-04-29:
OK...   I'll watch it now!

Rhonal89, 2017-04-29:
I appreciate all the work you put into marlin. Im loving ubl

Roxy-3D, 2017-04-29:
I agree with your analysis....    I'll get this annoy problem fixed ASAP!    When you are printing and you see the retraction happen...    Does it only do the retraction with nozzle going in one direction?   Or does it do it with the nozzle going both directions?

Rhonal89, 2017-04-29:
It looks like it does it in one direction. Can't tell clearly

Roxy-3D, 2017-04-30:
@Rhonal89   I think I might have found the problem...   Can you go into ubl_motion.cpp and change these lines:
```cpp
    const bool inf_normalized_flag = NEAR_ZERO(on_axis_distance),
               inf_m_flag = NEAR_ZERO(dx);
```
to this:
```cpp
    bool inf_normalized_flag=false, inf_m_flag=false; 

    inf_normalized_flag = isinf(e_normalized_dist);
    inf_m_flag = isinf(m);
```
The original logic got changed in a botched attempt to compress the code with ternary operators.
As soon as you can check it out...  I would appreciate feed back on whether or not it fixed the problem because I would like to have this code 100% functional for the promotion to 'Golden Status'.

Rhonal89, 2017-04-30:
Hi roxy, I tried the changes and it keeps happening.

Roxy-3D, 2017-04-30:
OK...  I'll keep searching...

Roxy-3D, 2017-05-01:
@Rhonal89    I can duplicate the problem.  And I have it narrowed down to just a few lines of GCode.  I've been adding debug code and looking at various values during the calculations...   I'm not there yet.  But I'm making slow and steady progress.

Rhonal89, 2017-05-01:
Yes, yes. Thank You!! Roxy. I been thinking and looking.  Where can I support you and the marlin team for all the hard work you guys are doing in getting marlin to were it currently is.

thinkyhead, 2017-05-01:
Keep me posted. Where does the issue seem to be centralized? Remember that Marlin has no concept of "layers." It just sees a Z move.

Roxy-3D, 2017-05-01:
> Keep me posted. Where does the issue seem to be centralized? Remember that Marlin has no concept of "layers." It just sees a Z move.

The problem is the extruder motor is actually going the wrong direction for part of the move in certain locations on the bed.   And it seems to be tied to crossing of mesh lines in certain directions.   The big problem is it hardly ever happens.   I'm not even sure it is happening in the mesh_buffer_line()  or ubl_motion.cpp file.   

But I've got a couple lines of gcode isolated that show the behavior.   I'm trying to get macros defined in PronterFace so I can go back and forth and watch it happen....  And once that is fully confirmed...  I'll be adding lots of debug information and figure out why the stepper motor is going the wrong direction.

There is a chance the problem is down in the stepper motor code...  But I'll know more by the end of the day.

Roxy-3D, 2017-05-01:
@Rhonal89    If you can...  Can you replace ubl_motion.cpp with this?   I think it is working for me.   I would like to hear if it works for you before I prepare a Pull Request and merge it.
```cpp
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#include "MarlinConfig.h"

#if ENABLED(AUTO_BED_LEVELING_UBL)

  #include "Marlin.h"
  #include "ubl.h"
  #include "planner.h"
  #include <avr/io.h>
  #include <math.h>

  extern float destination[XYZE];
  extern void set_current_to_destination();

  static void debug_echo_axis(const AxisEnum axis) {
    if (current_position[axis] == destination[axis])
      SERIAL_ECHOPGM("-------------");
    else
      SERIAL_ECHO_F(destination[X_AXIS], 6);
  }

  void debug_current_and_destination(const char *title) {

    // if the title message starts with a '!' it is so important, we are going to
    // ignore the status of the g26_debug_flag
    if (*title != '!' && !ubl.g26_debug_flag) return;

    const float de = destination[E_AXIS] - current_position[E_AXIS];

    if (de == 0.0) return;

    const float dx = current_position[X_AXIS] - destination[X_AXIS],
                dy = current_position[Y_AXIS] - destination[Y_AXIS],
                xy_dist = HYPOT(dx, dy);

    if (xy_dist == 0.0) {
      return;
      //SERIAL_ECHOPGM("   FPMM=");
      //const float fpmm = de / xy_dist;
      //SERIAL_PROTOCOL_F(fpmm, 6);
    }
    else {
      SERIAL_ECHOPGM("   fpmm=");
      const float fpmm = de / xy_dist;
      SERIAL_ECHO_F(fpmm, 6);
    }

    SERIAL_ECHOPGM("    current=( ");
[... 384 lines cut ...]
          e_position = start[E_AXIS] + on_axis_distance * e_normalized_dist;
          z_position = start[Z_AXIS] + on_axis_distance * z_normalized_dist;
        }
        else {
          e_position = end[E_AXIS];
          z_position = end[Z_AXIS];
        }

        planner.buffer_line(next_mesh_line_x, y, z_position + z0 + ubl.state.z_offset, e_position, feed_rate, extruder);
        current_xi += dxi;
        xi_cnt--;
      }
    }

    if (ubl.g26_debug_flag)
      debug_current_and_destination(PSTR("generic move done in ubl_line_to_destination()"));

    if (current_position[X_AXIS] != end[X_AXIS] || current_position[Y_AXIS] != end[Y_AXIS])
      goto FINAL_MOVE;

    set_current_to_destination();
  }

#endif
```

Rhonal89, 2017-05-02:
Hi Roxy, Here is a video https://www.youtube.com/watch?v=i477Xxrcv7k

Roxy-3D, 2017-05-02:
Thanks!!!!  I'll watch again in the morning...   I was able to print the test print with rectangles at different angles....   It is very possible I don't have things right.   But I do understand what is wrong....

Rhonal89, 2017-05-02:
Thank You!! Roxy. I was just looking at the code wishing I knew what I was looking at.

Roxy-3D, 2017-05-02:
I'm using that code right now to do a horribly complicated print.   I'll post pictures in the morning.   Can you grab the very most current RCBugFix and get that up and running on your printer?   Because the Pull Request I do will be against that.   I'm pretty sure I've got that bug you found in the other video fixed.

Roxy-3D, 2017-05-02:
https://github.com/MarlinFirmware/Marlin/pull/6540

Rhonal89, 2017-05-04:
Hi Roxy, I updated to the latest marlin rc8fix and it's printing great. Now to tune ubl some more. Thank You!! For all the help.

Roxy-3D, 2017-05-04:
You wouldn't believe how hard that bug was to find.   UBL has some optimizations in the calculations for the Z-Height correction.   When the nozzle is moving in a purely horizontal or vertical direction, you know you are only crossing X-Mesh lines or Y-Mesh lines.   So things can be made easier on the CPU.

But...  That doesn't mean the nozzle is moving horizontal or vertical.  It just means the nozzle is only crossing mesh lines in one direction or the other.     The case we had was 'horizontal movement' from one mesh cell to the next.  But the actual movement was mostly vertical.    And the wrong optimization code was assuming it could use the wrong number....    Aaaauuughhhhh!

I think that one is behind us now!   Thank You for the help!

Rhonal89, 2017-05-04:
I bet it took lots of work. And thanks for the explanation I get it better and Thank You!! for all your hard work Roxy.

closed 2017-05-04 by Rhonal89