MarlinFirmware/Marlin #6479: BLTouch heater does not come back on

oderwat, 2017-04-28:
After the newest commit 6e8ecb908ad0585f895510c4f67e9579cc6d1710 using G28 and G29 with AUTO_BED_LEVELING_LINEAR the heaters didn't came back on after G29 and also not after G28.

I tried to find a reason and it has to do with my code which resets the BLTouch before every print (to make sure it is in a working state):

```
G28 X0 Y0

M280 P0 S160 ; Reset 3D-Touch
G4 P500
M280 P0 S10 ; Pin out
G4 P500
M280 P0 S90 ; Pin back
G4 P500

M104 S[extruder0_temperature] T0
M190 S[bed0_temperature] T0
M109 S[extruder0_temperature] T0
G28 Z0
```

After removing the code it works. I did not check the details but I guess the BLTouch cycle counter gets screwed up by above code. I think that needs to be fixed.

Roxy-3D, 2017-04-28:
I made changes to give you control of the behavior.  If you comment out:  //#define BLTOUCH_HEATERS_OFF  you will have the old behavior that you prefer.
```cpp
//#define BLTOUCH
//#define BLTOUCH_DELAY 375      // (ms) Enable and increase if needed
//#define BLTOUCH_HEATERS_OFF // if defined the printer's heaters are
turned off during probe event
```
Please check out:  https://github.com/MarlinFirmware/Marlin/pull/6485    It will be merged as soon as Travis says everything is OK.

oderwat, 2017-04-28:
I don't prefer it not doing the switching :) Actually I did switch bed and nozzle of myself for other sensors. What I would prefer is that it works as expected even when I use M280 for Reset / Test / Pin up / Pin down. I think that starting G28 and G29 or using any G280 command related to the BLTouch should reset the cycle counter or give me a possibility to reset it with a command after testing procedures.

Roxy-3D, 2017-04-28:
> I think that starting G28 and G29 or using any G280 command related to the BLTouch should reset the cycle counter or give me a possibility to reset it with a command after testing procedures.

OK...   I understand.   But those are too invasive to be making at this point in time.  It is very likely something else would get broken.   I will circle back and figure out how to do this for you once v1.1.0 gets tagged.

oderwat, 2017-04-28:
OK. I think you mean you won't change code inside G28 or G29 (like if BLTouch cycle count = 0 ...)? What about changing it in a way such that the cycle count is not affected when the BLTouch statue is changed while the heaters are off? That way my gcode's M280 commands would simply not change the cycle counter at all as it does not matter when heaters are off anyway?

Roxy-3D, 2017-04-28:
Well...   G28 and G29 are very complicated with all of the conditional code.   And they both use things with lots of conditional code like:
```cpp
static void do_homing_move(const AxisEnum axis, float distance, float fr_mm_s=0.0) 
static float run_z_probe() 
static void homeaxis(const AxisEnum axis)
etc.
```
To do the BL-Touch heater modification I only needed to cut into the logic in two places...   That is why I felt I could do and not cause a problem.   And even so...   You found a use case I didn't anticipate.   It is more that thinking that is keeping me from doing anything right now.

But even so...   Once v1.1.0 gets tagged...  The number of changes to accomplish what you want are not going to be that bad.   My preference would be those changes only exist in the development branch but even if I get my way, it will be almost trivial for you to cross those few lines of code over to the official v1.1.0 branch if you want be running the Stable Release.    

Also...   The Inductive Probe people are now asking "Will this help us?"   I suspect the answer is "Yes!" so if that is the case, a more comprehensive solution will be coming soon.

Grogyan, 2017-04-28:
I've posted the link to the BLTouch community, to see if it really makes a difference overall,l turning off the heaters, or whether it's a specific case by case where it will make a difference. 

Though with the latest commit, the feature can be disabled.

oderwat, 2017-04-28:
I think that it is even more important with other sensors. I analysed a lot off z-offset problems and I know from this that you generally get more stable results if you switch of both heaters when doing the z-offset measurements. This may have different effects with different sensors. So I think it is pretty valid to assume that having it switched of while probing is "best" for all kind of sensors. This would indeed need complete different changes in the code. Those may be much easier to reason about than having side effects in the BLTouch code.

Roxy-3D, 2017-04-29:
> This would indeed need complete different changes in the code. Those may be much easier to reason about than having side effects in the BLTouch code.

Or...  A simpler way to deal with it is....   Just turn the heat of around ANY probe reading.   That takes care of all type of sensors including micro-switches which don't need it.  But the micro-switches aren't going to complain if you turn the heaters off when you are probing a point.  That reduces a huge amount of corner cases.

oderwat, 2017-04-29:
Yes, that is what I was suggesting. Wrap all probing with an "heater off -> probe -> heater on". Of course only if the heater was on in the first place.

closed 2017-11-01 by thinkyhead