MarlinFirmware/Marlin #6480: RepRap Discount FullGraphicDisplay XYZ position numbers glitch

oderwat, 2017-04-28:
Since one of the latest updates the display of my RepRap Discount Full Graphic Display shows small and negative numbers wrong:

`X3-33` and `X1 20` instead of `X -33` and `X  20` for example. Same for Y and Z.

benlye, 2017-04-28:
Something to do with the changes in utlity.cpp in https://github.com/MarlinFirmware/Marlin/commit/87d240042aeaad6772743e93da17c657a60878f8?  Will see if I get the same on mine later on.

benlye, 2017-04-28:
In itostr4sign, the first char in the string isn't blanked out if the string is less than three chars long.

Change this:
```
      else {
        conv[4] = ' ';
        if (xx >= 10) {
          conv[4] = neg ? '-' : ' ';
          conv[5] = DIGIMOD(xx, 10);
        }
        else {
          conv[4] = ' ';
          conv[5] = neg ? '-' : ' ';
        }
      }
```

To this:
```
      else {
        conv[3] = ' ';
        conv[4] = ' ';
        if (xx >= 10) {
          conv[4] = neg ? '-' : ' ';
          conv[5] = DIGIMOD(xx, 10);
        }
        else {
          conv[5] = neg ? '-' : ' ';
        }
      }
```

PR on the way.

oderwat, 2017-04-28:
Is fixed for me 👍

closed 2017-04-28 by oderwat