MarlinFirmware/Marlin #6483: How to change configuration file

batata004, 2017-04-28:
Hi,

With a lot of effort I could update my old Marlin configuration file (provided by my printer manufacturer) to the newest one.  Attached are my 2 conf files.

Nowadays I am using a servo (connected to an arm with a switch) to probe the surface of my heated bed (glass). Today I will be changing this servo to an inductive probe and I will also be changing the glass to aluminium. So my question is this: which configs should I change in order to disable the servo probing and enable the inductor probing?

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/964397/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/964398/Configuration_adv.txt)

Bob-the-Kuhn, 2017-04-28:
Below are the changes that should get you up & running.  

All the interface circuits I've seen use a fairly high impedance resistor network so the pullup for the Z_MIN needs to be disabled.

```
#define FIX_MOUNTED_PROBE

//#define Z_ENDSTOP_SERVO_NR 0
//#define Z_SERVO_ANGLES {90,0} // Z Servo Deploy and Stow angles

//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

```
I don't remember if the inverting flags should be true or false
```
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
```

batata004, 2017-04-28:
Thanks man, I am glad this marlin community is very active and helpful otherwise it wouldnt be easy to configure this firmware.

I THINK there should be some sort of visual tool where the user can go step by step answering questions like "Do you use any probe?"   "Is it a fixed probe?"    "Is it a servo probe?"... This way it would be much easier the adoption of this great firmware. I think if it's not possible to create a visual tool, it should be some documentation where the user can go one by one and make sure he/she configured the right stuff.

closed 2017-04-28 by batata004

batata004, 2017-05-04:
@Bob-the-Kuhn i did exactly the changes you said and I am using a voltage divider in my sensor but for some reason the LED still does not turn on (the builtin led of the sensor).

A guy told me it could be a problem with the pullup resistor, are you certain about the section of pullup you provided? Should I really do that? In my original code it was like this:

```
// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif
```