MarlinFirmware/Marlin #6486: Feature request

batata004, 2017-04-28:
Hi my friends,

Today I was thinking this "I am not using 2 extruders and because of that my RAMPS 1.4 has one spare MOSFET that could be used to run a second extruder that I dont have."

Since I have this "spare mosfet" I would like to use it to turn on/off my LED stripe (that I use to light up my printer), the cooler that spins over the RAMPS and the cooler that runs over my extruder.

In my current setup, I am connecting the LED stripe and both cooler to the 12V from my power supply and because of that they are permantenly on. So I was wondering if I could turn them on before every print and turn them off after every print.

Is there a feature in marlin firmware that allows me to do that? Sometimes I ask my printer to print something that takes 5 or 6 hours, so I go to sleep. Next day when I wake up my part is printed BUT leds and coolers remain spinning for the whole night, even after the printing is finished. It would be nice to be able to turn those things off without any extra hardware since ramps already has enought of it.

Kaibob2, 2017-04-28:
You can just  figure out which Pin the Mosfet is connected to and then switch this Pin with M42
http://reprap.org/wiki/G-code#M42:_Switch_I.2FO_pin
I suppose the Pin will be PWMable so you can change brightness of your Led strip. 
For example add M42 P7 S255 in your slicers startcode will switch  Pin7 to full on. 
M42 P7 S0 in your slicers end code will switch Pin7 off. 
M42 P7 S127 will switch the LED  strip to 50% brightness.

Bob-the-Kuhn, 2017-04-28:
Does the extruder fan already operate this way?  If yes then just add these items in parallel with it.

Which of your FETs is unused?  If it's the FAN FET then see if M106 & M107 will control it.
```
  /**
   * M106: Set Fan Speed
   *
   *  S<int>   Speed between 0-255
   *  P<index> Fan index, if more than one fan
   */
```

If the above doesn't take care of it then you'll need use @Kaibob2's method.  In order to use the M42 method you'll need to get the FET off the protected pins list.  To do that you'll need to comment out a line in the pins_RAMPS.h file.

If the FAN pin isn't used then the command you'll use is M42 P9 Sxxx and you'll need to comment out line 166.

If the heated bed isn't used then the command you'll use is M42 P8 Sxxx and you'll need to comment out line 167.
```
line 165    #if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
line 166      #define FAN_PIN        RAMPS_D9_PIN
line 167      #define HEATER_BED_PIN RAMPS_D8_PIN
```

Tannoo, 2017-04-29:
For rgbs in RCBugfix:
```
/**
 * RGB LED / LED Strip Control
 *
 * Enable support for an RGB LED connected to 5V digital pins, or
 * an RGB Strip connected to MOSFETs controlled by digital pins.
 *
 * Adds the M150 command to set the LED (or LED strip) color.
 * If pins are PWM capable (e.g., 4, 5, 6, 11) then a range of
 * luminance values can be set from 0 to 255.
 *
 * *** CAUTION ***
 *  LED Strips require a MOFSET Chip between PWM lines and LEDs,
 *  as the Arduino cannot handle the current the LEDs will require.
 *  Failure to follow this precaution can destroy your Arduino!
 * *** CAUTION ***
 *
 */
//#define RGB_LED
//#define RGBW_LED
#if ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define RGB_LED_R_PIN 34
  #define RGB_LED_G_PIN 43
  #define RGB_LED_B_PIN 35
  #define RGB_LED_W_PIN -1
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ENABLED(BLINKM) || ENABLED(RGB_LED) || ENABLED(RGBW_LED)
  #define PRINTER_EVENT_LEDS
#endif
```

Tannoo, 2017-04-29:
It should be possible to control a single color strip just as well. Although, there is a case_light feature for single lights that is on/off control with `M350`.

Tannoo, 2017-04-29:
>In my current setup, I am connecting the LED stripe and both cooler to the 12V from my power supply and because of that they are permantenly on. So I was wondering if I could turn them on before every print and turn them off after every print.

With the case_light feature, yes.

batata004, 2017-04-29:
Hi my friends,

Actually I dont wanna know how to control this LED stripe, I wanna connect 2 other fans in parallel with it AND when my LED stripe gets on everything gets on.

In my RAMPS I have the 2 pins at 9 free. See this image please: https://goo.gl/photos/Sxoc2EnDBTkRrkpK8

The 8 and 10 I am already using (hotend and heated bed). So how can I use the 2 pins at 9 (GND and 12v) to turn on/off LED + 2 FANS? 

I mean, if I connect in parallel the LED stripe and 2 fans into these 2 pins, how do I turn it on and off before everyprint?

Kaibob2, 2017-04-29:
Connect the fans and the stripe to the header. 
Add M42 P9 S255 in your slicers startcode will switch the stripe and fans on.
M42 P9 S0 in your slicers end code will switch the fans and stripe off.

batata004, 2017-04-29:
@Kaibob2 THANKS!!! I just read the docs and you are damm right, it should work and I will do that right now! Just one question: how did you know it's the pin 9 of arduino MEGA that this mosfet (that I have unused) is connected of my ramps 1.4?

Kaibob2, 2017-04-29:
You wrote "In my RAMPS I have the 2 pins at 9 free."

batata004, 2017-04-29:
So you are not sure if that 9 labeled at RAMPS 1.4 means actually the pin 9 of arduino mega?

Tannoo, 2017-04-29:
[image]

vMeph, 2017-09-24:
Just one question from what i understand pin 11 pwm is used by a timer already?
If pin 11 is used by one of the rgb will that interfer with the print events led order?

closed 2018-02-09 by thinkyhead