MarlinFirmware/Marlin #6488: Z-Babystepping Issue

jasoncross, 2017-04-28:
I have the following in my advanced config:

#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
#define BABYSTEP_XY // Also enable X/Y Babystepping. Not supported on DELTA!
#define BABYSTEP_INVERT_Z false // Change if Z babysteps should go the other way
#define BABYSTEP_MULTIPLICATOR 1 // Babysteps are very small. Increase for faster motion.
#define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping
#define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
#define DOUBLECLICK_MAX_INTERVAL 1250 // Maximum interval between clicks, in milliseconds.
// Note: Extra time may be added to mitigate controller latency.
#endif

When my first layer is printing, I double-click the wheel and the screen changes to Z-Offset.  I rotate the wheel, the number changes, but nothing changes with my Z as it's printing.  For example, my default was -1.4.  I turned it all the way to -.4 with no difference - yet 1mm should be a HUGE difference while printing.

I've also tried with the printer idle - my Z isn't moving as I turn the encoder.

Running the RCBugfix version from two days ago with UBL enabled.

Roxy-3D, 2017-04-28:
I don't have an answer for what you describe while printing...    But I did verify what you see with an RCBugFix from two days ago running UBL.

If I power up the printer...   And drop into Z-BabyStepping with a Double Click of the Encoder Wheel...  The system will not adjust the Z-Height.    However...   Once I do a G28, Z-BabyStepping works as expected.   It would seem to me Z-BabyStepping should work even with the printer in a non-homed state.

Roxy-3D, 2017-04-28:
This is not a UBL issue.   I just tried it with all bed leveling turned off and the same thing happens.

But...   That is what the code is intending to happen:
```cpp
    #if ENABLED(BABYSTEPPING)
      static void babystep_axis(const AxisEnum axis, const int distance) {
        if (axis_known_position[axis]) {
          #if IS_CORE
            #if ENABLED(BABYSTEP_XY)
              switch (axis) {
                case CORE_AXIS_1: // X on CoreXY and CoreXZ, Y on CoreYZ
                  babystepsTodo[CORE_AXIS_1] += distance * 2;
                  babystepsTodo[CORE_AXIS_2] += distance * 2;
                  break;
                case CORE_AXIS_2: // Y on CoreXY, Z on CoreXZ and CoreYZ
                  babystepsTodo[CORE_AXIS_1] += CORESIGN(distance * 2);
                  babystepsTodo[CORE_AXIS_2] -= CORESIGN(distance * 2);
                  break;
                case NORMAL_AXIS: // Z on CoreXY, Y on CoreXZ, X on CoreYZ
                  babystepsTodo[NORMAL_AXIS] += distance;
                  break;
              }
            #elif CORE_IS_XZ || CORE_IS_YZ
              // Only Z stepping needs to be handled here
              babystepsTodo[CORE_AXIS_1] += CORESIGN(distance * 2);
              babystepsTodo[CORE_AXIS_2] -= CORESIGN(distance * 2);
            #else
              babystepsTodo[Z_AXIS] += distance;
            #endif
          #else
            babystepsTodo[axis] += distance;
          #endif
        }
      }
    #endif // BABYSTEPPING
```
The author of the code very clearly did not want baby stepping to happen on an axis unless the axis had been homed.    Probably...  That was done for safety and to protect the machine.   But, the baby steps are under the user's control..   So I would be OK with relaxing that constraint.   (At least right now, with what I know...)

jasoncross, 2017-04-28:
I just did a G28 and then went into baby stepping and still it didn't actually change my z-axis.

@Roxy-3D Do you have #define BABYSTEP_ZPROBE_OFFSET enabled?

Roxy-3D, 2017-04-28:
No.   Actually...   I'm kind of scared of that option.   I didn't notice that when you posted your Configuration.h settings.   If I had noticed that, I would have requested you turn that off.

The problem with that option in UBL is I have zero test time on what happens when the zprobe_zoffset number shifts.    And in fact, when somebody is trying to bring up UBL, I urge them as a very first step to get their Z_PROBE_OFFSET_FROM_EXTRUDER number very accurately set because that helps things later.

jasoncross, 2017-04-28:
I wonder if it's an issue when that is set.  Will turn it off and test when I can.

jasoncross, 2017-04-28:
Confirmed!  When Z_PROBE_OFFSET_FROM_EXTRUDER is disabled in firmware, double-clicking changes the screen to "babystepping" and it DOES move the Z-axis in real time.  When Z_PROBE_OFFSET_FROM_EXTRUDER is enabled, the screen changes to "Z-Axis Offset" and it changes that, but not the Z-level itself.  

Based on the comment in firmware of "// Enable to combine M851 and Babystepping" I'd say something is wrong there as I'd expect both to change.

Roxy-3D, 2017-04-28:
To fix this code before Sunday is much too complex.    For now...  Please just turn it off and we will circle back and try to figure out the right way to handle this after v1.1.0 gets tagged.

But until that time...  if you want to shift the mesh height up or down with UBL...  You can do a G29 P6 C #

Tannoo, 2017-04-29:
By design? #6362

Tannoo, 2017-04-29:
And before that, it was discussed here: #5887

jasoncross, 2017-04-29:
Neither of those explain why Z_PROBE_OFFSET_FROM_EXTRUDER being enabled is keeping babystepping from actually moving the z-axis.

Tannoo, 2017-04-29:
If it is homed, or knows it's location, it should work fine. It did the last time I used it.

Tannoo, 2017-04-29:
I can check it again in the morning.

jasoncross, 2017-04-29:
It's homed. It knows it's location. It's not moving the z-axis.

Tannoo, 2017-04-29:
I have found also that I had to set the multiplicator to 20 to get the movement I wanted.

jasoncross, 2017-04-29:
> I have found also that I had to set the multiplicator to 20 to get the movement I wanted.

Sure but as I said in the first post, changing it from -1.4 to -.4 (and further) should show a noticeable difference - multiplier or not. It's not happening. It changing the z-offset set but not actually moving the z-axis in realtime.

Tannoo, 2017-04-29:
It used to work just fine with the multiplicator at 4 for me. Something got changed and had to increase that.

It's moving, but with multiplicator at 1, you will not see it as the movement are likely around .00001 per encoder click.

Tannoo, 2017-04-29:
I will see what my printer does in the moring. Atm, I am at work.

jasoncross, 2017-04-29:
Cranking the wheel until it shows a new offset a full mm SHOULD BE noticeable regardless of how many clicks it takes. Plus as I also said in an earlier post, when the Z Probe offset option is disabled, I can tell it's moving.

Tannoo, 2017-04-29:
Yeah, I do have my inductive probe offset in all three directions.

jasoncross, 2017-04-29:
Sorry I meant if 

#define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping

Is set. Not if the z probe offset is set.

jasoncross, 2017-04-29:
Check this out:

      #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!no_babystep && planner.abl_enabled)
          thermalManager.babystep_axis(Z_AXIS, -lround(diff * planner.axis_steps_per_mm[Z_AXIS]));
      #else
        UNUSED(no_babystep);
      #endif

Are you sure planner.abl_enabled is being set to true with UBL?  Not finding in code where it is.

Tannoo, 2017-04-29:
In `ultralcd.cpp`:
```
    #if ENABLED(BABYSTEP_ZPROBE_OFFSET)

      void lcd_babystep_zoffset() {
        if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
        defer_return_to_status = true;
        ENCODER_DIRECTION_NORMAL();
        if (encoderPosition) {
          const int babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
          encoderPosition = 0;

          const float new_zoffset = zprobe_zoffset + planner.steps_to_mm[Z_AXIS] * babystep_increment;
          if (WITHIN(new_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX)) {

            //if (planner.abl_enabled)    <<-----Commenting this out allows babystepping when not printing.
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);

            zprobe_zoffset = new_zoffset;
            refresh_zprobe_zoffset(true);
            lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
          }
        }
        if (lcdDrawUpdate)
          lcd_implementation_drawedit(PSTR(MSG_ZPROBE_ZOFFSET), ftostr43sign(zprobe_zoffset));
      }
```

Tannoo, 2017-04-29:
Maybe someone else will find the "real" fix.

jasoncross, 2017-04-29:
I have confirmed that this issue is a combination of UBL and BABYSTEP_ZPROBE_OFFSET.  

The root cause appears to be that when BABYSTEP_ZPROBE_OFFSET is set, the axis isn't allowed to move unless planner.abl_enabled is set to true.

In ultralcd.cpp, I changed 

if (planner.abl_enabled)
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);

to

if (planner.abl_enabled || ENABLED(AUTO_BED_LEVELING_UBL))
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);

And in Marlin_main.cpp, I changed:

#if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!no_babystep && planner.abl_enabled)
          thermalManager.babystep_axis(Z_AXIS, -lround(diff * planner.axis_steps_per_mm[Z_AXIS]));
      #else
        UNUSED(no_babystep);
      #endif

to 

#if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!no_babystep && (planner.abl_enabled || ENABLED(AUTO_BED_LEVELING_UBL)))
          thermalManager.babystep_axis(Z_AXIS, -lround(diff * planner.axis_steps_per_mm[Z_AXIS]));
      #else
        UNUSED(no_babystep);
      #endif

And now it moves the axis while babystepping with z-offset while using UBL.

A likely better fix would be to update the code in Marlin_main.cpp to set planner.abl_enabled to true when using UBL (as it does now with other leveling) but I don't know if that was omitted intentionally due to some other issue or if it was an oversight.

jasoncross, 2017-04-29:
Doing it my way above throws a scope error when compiling without UBL, so it looks like planner.abl_enabled may be the only way to go.

tcm0116, 2017-05-04:
> The author of the code very clearly did not want baby stepping to happen on an axis unless the axis had been homed.

As the author of that change, I can confirm that. On my machine, trying to babystep a motor while it's disabled results in a somewhat violent movement as the motor is enabled, stepped, and then disabled for each babystep. 

> I have confirmed that this issue is a combination of UBL and BABYSTEP_ZPROBE_OFFSET.

These two features were developed independently, so this make sense.

> Doing it my way above throws a scope error when compiling without UBL

The following should work in `Marlin_main.cpp` with a similar approach being used in `ultralcd.cpp`:

      #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        if (!no_babystep &&
          #if ENABLED(AUTO_BED_LEVELING_UBL)
            ubl.enabled // TODO: Figure out what variable this really should be
          #else
            planner.abl_enabled
          #endif
          )
          thermalManager.babystep_axis(Z_AXIS, -lround(diff * planner.axis_steps_per_mm[Z_AXIS]));
      #else
        UNUSED(no_babystep);
      #endif

Roxy-3D, 2017-05-04:
>> The author of the code very clearly did not want baby stepping to happen on an axis unless the axis had been homed.

> As the author of that change, I can confirm that. On my machine, trying to babystep a motor while it's disabled results in a somewhat violent movement as the motor is enabled, stepped, and then disabled for each babystep.

Hey!!!!   THANK YOU FOR YOUR BABYSTEP WORK!   I love Z-BabyStepping!!!!!!

tcm0116, 2017-05-04:
>Hey!!!! THANK YOU FOR YOUR BABYSTEP WORK! I love Z-BabyStepping!!!!!! 

Just to clarify, I worked on the BABYSTEP_ZPROBE_OFFSET feature and also added the `if (axis_known_position[axis])` one-liner 😁

closed 2017-11-01 by thinkyhead