MarlinFirmware/Marlin #6489: MESH leveling anomalous behavior.

WheresWaldo, 2017-04-28:
Using RCBugFix downloaded on 4/27/2017 and configuring for MESH bed leveling I have found the following anomalies:

1. Bed leveling starts each point at 0.200 according to the Full Graphics LCD Smart Controller. I would have expected it to start at 0.000
2. G29 S4 Zn.nnn does not appear to do anything with respect to offsetting the MESH. Has it been deprecated in the latest builds.

My bed is warped when heated, I know that it is and I know by how much. When using G29 S4 there appears to be no change no matter what is placed in that parameter. What appears odd it that M851 seems to affect the MESH, when I put the same parameter in M851 the MESH is now offset. 

I am just trying to figure out if I have missed something and is M851 going to be the Z offset used for all leveling schemes.

Grogyan, 2017-04-29:
For me I see the nozzle offset not applied to the mesh. 
So if I have a nozzle offset of -2mm
Printing at 0.2mm
See printing at 1.2mm

Can't really tell to well by my eye, but it is printing right now about 1.2 mm above the bed, not the 0.2 it is meant to be

Roxy-3D, 2017-04-29:
Do you have:  #define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping
turned on?   I think we have confirmed it doesn't work with UBL.   https://github.com/MarlinFirmware/Marlin/issues/6488#issuecomment-298120988

WheresWaldo, 2017-04-29:
@Roxy-3D 
Nope, all that is enabled in babysteps are:

1. babystepping (of course)
2. BABYSTEP_INVERT_Z false
3. BABYSTEP_MULTIPLICATOR

Please note that this is MESH not UBL!
MESH simply ignores G29 S4 Zn.nn I put Z as high as 2.0 mm and still tried to extrude smashed to the bed. Then I zeroed out G29 S4, saved to EEPROM then enabled M851 with the same offset and gues what, it offset the MESH. It isn't supposed to work that way, it isn't documented that way, it didn't work that way in the past. Unfortunately I cannot go back and see exactly when it was broken so I can't even begin to tell you where to look. Also point 1 in the original post is behavior not seen before. When each new probe point is displayed, it starts with a zero then jumps to 0.200 when the LCD refreshes with no input whatsoever.

Tannoo, 2017-04-29:
You might want to up `BABYSTEP_MULTIPLICATOR` from the 1 to something more to your liking. I moved mine to 20, but you may not want it that high.

Tannoo, 2017-04-29:
You also may need to restore EEPROM defaults and then save again.

WheresWaldo, 2017-04-29:
Multiplicator is not the issue, nor is clearing or saving the EEPROM.

jaysonkelly, 2017-05-02:
@WheresWaldo I think I found where the 0.200 offset is coming from. There are two move to next point functions which now have a + 0.2 mm on the z position before moving to each leveling point. _manual_goto_xy and _manual_probe_goto_xy are the two functions. Commenting out the +0.2 on the z position modifier line removes the offset. It looks like they are trying to give some room for movement on each of the leveling points. 

Now I need to find out why the move screen isn't displaying. My screen goes straight to the get z screen when moving to each leveling point.

thinkyhead, 2017-05-03:
> Bed leveling starts each probe point at 0.200

That is intentional.

> MESH simply ignores G29 S4 Zn.nn 

Well, the `gcode_G29` function _does_ set `mbl.z_offset` to the value provided.
```cpp
        if (code_seen('Z')) {
          mbl.z_offset = code_value_linear_units();
        }
```
And the `mesh_bed_leveling::get_z` function does add it to the value it returns.
```cpp
      return z_offset + z0
```
And, that's as complicated as it gets. Calling `mbl.reset()` clears `mbl.z_offset` (so it is also cleared by `G29 S1`.)

WheresWaldo, 2017-05-03:
Maybe that is what it is supposed to do but I just checked yesterday's code again and it still ignores G29 S4 Zn.nnn to set mesh offset. Just to be clear here are the steps I did

1. Download current zip
2. Enable all required defines for MESH, SDSUPPORT, FG LCD, Proper number of steps, fixed probe, safe homing, babysteps (but not babystep to offset)
3. compile and flash
4. M502 followed by M500 which should clear everything
5. On the LCD Prepare >> Level Bed
6. Since my probes trigger below Z0 I need to move every point up well beyond 0.20 mm, it would be nice if this wasn't arbitrary.
7. Save mesh with M500 (I am pretty sure Marlin already saves the mesh but this makes me feel better)
8. Print a 10mm x 10mm x 2mm test cube and measure it's thickness when done, whatever difference becomes the Z_Offset usually set with G29 S4 Zn.nnn which seems to not work but now set with M851 which appears to work.

At no point after that is G29 S1 ever called. Not from the terminal, not in startup GCode.

I would try UBL if it was actually easier to use with manual probing and didn't shift the UBL mesh to the right by on grid spacing. The other reason that MESH done fully manually works better is the poor implementation of this printer's min Z axis endstops and the amount of flex in the build plate. It uses the nozzle tip as the probe and lifts the carriage off the min Z limit switches after it hits the bed. The bed flexes because it is only held in the extreme four corners and the weight of the hotend extruder is completely on the probe point at each probing location. The flex is inconsequential when actually printing as the nozzle in that case is always above the bed. That bed flex makes UBL almost worthless on this printer implementation since I have to manually adjust every single probe point in the middle of the bed and adjusting ain't easy.

Roxy-3D, 2017-05-03:
> I would try UBL if it was actually easier to use with manual probing and didn't shift the UBL mesh to the right by one section of grid spacing. 

More detail would be helpful.   I don't know what you mean about shifting the UBL mesh to the right???

> That bed flex makes UBL almost worthless on this printer implementation since I have to manually adjust every single probe point in the middle of the bed and adjusting ain't easy. 

You could position the nozzle over the center of the bed and just fill the values in.  Maybe something like G29 P3 R5 C #   to do 5 at a time.       But the manual mesh editing of the center points of the bed is going to take much less time than doing a manual probe there.   Using the editor to bring those points up should be very quick.

WheresWaldo, 2017-05-03:
Sorry @Roxy-3D Since I don't use UBL currently I do not feel qualified to open a ticket I only tested it one time (literally) and went immediately to MESH which I am familiar with. Here is what I saw, the UBL automatic routine skipped the whole first column of Y probe points then tried to go beyond X Max on the other end.

Maybe I have a fundamental misunderstanding about editing the mesh in UBL but It certainly is easy to just manually do a mesh plane with MESH leveling, If UBL was as easy I would make sure every Robo3D R1 user on their forums currently using Marlin 1.1.0 would use UBL. All the Marlin 1.1.0 users on the R1 are using configuration files that I am providing to that community.

Sorry very far off topic for this issue.

WheresWaldo, 2017-05-04:
@Roxy-3D I know that you are an extremely busy person but I would like to chat with you for a few moments about UBL if you are willing please contact me directly, bud.hammerton at gmail.com

closed 2017-11-01 by thinkyhead