MarlinFirmware/Marlin #6496: Dual X Carriage Positioning?

kcheeeung, 2017-04-29:
Hey everyone again. It's me.

I want to understand more about the Dual X Carriage in the advanced settings. It seems pretty easy to setup. Is the Marlin dual x carriage only configured for have 2 extruders? One on each side?

Apparently I compiled with 4 extruders and the dual x carriage and that works. What are the orientations for these extruders? I also see there is a section for the offsets.

kcheeeung, 2017-05-03:
Hello everyone I rephrased my question.

thinkyhead, 2017-05-03:
5 extruders are supported. However, the special modes of Dual X assume the first extruder is alone, and all other extruders are on the 2nd carriage.

kcheeeung, 2017-05-07:
Oh okay. Would you be able to show me how to modify the code such that T0 and T1 are on the first carriage, then the rest be on the second (even if you hardcode).

I looked in planner.cpp and stepper_indirection.h and it appears that this Dual X feature controls the steppers here.

closed 2017-06-17 by kcheeeung