MarlinFirmware/Marlin #6497: Automatic power on (M80)

blole, 2017-04-29:
So when I came to Marlin, I saw that there was support for using the PS_ON feature of ATX power supplies and immediately installed such a supply on my printer so the noisy fans would only run when needed.

But when trying out different slicers and OctoPrint, that means having to add the M80 and M81 commands in the pre/post-gcode. Most recently it came up when using the pre-heat bed button in the OctoPrintPlugin for Cura (fieldOfView/OctoPrintPlugin#35). It also trips me up when controlling the printer manually from the LCD during calibration.

From [Configuration.h](https://github.com/MarlinFirmware/Marlin/blob/6424068412505b942ea81e6aaeac39499ef819ea/Marlin/Configuration.h#L184-L199)
```c++
/**
 * Select your power supply here. Use 0 if you haven't connected the PS_ON_PIN
 *
 * 0 = No Power Switch
 * 1 = ATX
 * 2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)
 *
 * :{ 0:'No power switch', 1:'ATX', 2:'X-Box 360' }
 */
#define POWER_SUPPLY 0

#if POWER_SUPPLY > 0
  // Enable this option to leave the PSU off at startup.
  // Power to steppers and heaters will need to be turned on with M80.
  //#define PS_DEFAULT_OFF
#endif
```
I think it could be useful to have an option like `#define AUTO_POWER` here that turns on the power before any bed/hotend/stepper/fan commands.

I don't know if there would be any use having more granularity than that, e.g. `AUTO_POWER_BED`, `AUTO_POWER_STEPPERS`; in my case that's not needed anyway since they all need the power to be on.

Tannoo, 2017-04-29:
Like this? #5875

Tannoo, 2017-04-29:
Search for `POWER` and you'll find a few discussion and PR's about it.

boelle, 2019-02-18:
@blole i think this has been added, try latest bugfix 2.0

boelle, 2019-03-06:
@thinkyhead i think we can close this one

blole, 2019-03-11:
Ah yes, I tried the latest bugfix-2.0.x now and saw the `AUTO_POWER_CONTROL` option that solves this. Thank you.

closed 2019-03-11 by blole