MarlinFirmware/Marlin #6502: #define INVERT_E1_DIR True not fefined

pmbroth, 2017-04-30:
I have this in my configuration.h file
// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR false
#define INVERT_E1_DIR True
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false

But I get the following errors related to thwe compiler not seeing the E1 being set as True
Using RCBUGFIX as of 4/30/17

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\stepper.cpp:47:

sketch\stepper.cpp: In static member function 'static void Stepper::set_directions()':

Configuration.h:759: error: 'True' was not declared in this scope

 #define INVERT_E1_DIR True

                       ^

sketch\fastio.h:49:36: note: in definition of macro '_WRITE_C'

 #define _WRITE_C(IO, v)   do { if (v) { \

                                    ^

sketch\fastio.h:74:21: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V) _WRITE(IO,V)

                     ^

sketch\stepper_indirection.h:315:31: note: in expansion of macro 'WRITE'

   #define E1_DIR_WRITE(STATE) WRITE(E1_DIR_PIN,STATE)

                               ^

sketch\stepper_indirection.h:442:108: note: in expansion of macro 'E1_DIR_WRITE'

     #define REV_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                                                                                                            ^

sketch\stepper_indirection.h:442:121: note: in expansion of macro 'INVERT_E1_DIR'

     #define REV_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                                                                                                                         ^

sketch\stepper.cpp:309:7: note: in expansion of macro 'REV_E_DIR'

       REV_E_DIR();

       ^

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\stepper.cpp:47:

Configuration.h:759: error: 'True' was not declared in this scope

 #define INVERT_E1_DIR True

                       ^

sketch\fastio.h:47:36: note: in definition of macro '_WRITE_NC'

 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= _BV(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~_BV(DIO ## IO ## _PIN); }; } while (0)

                                    ^

sketch\fastio.h:74:21: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V) _WRITE(IO,V)

                     ^

sketch\stepper_indirection.h:315:31: note: in expansion of macro 'WRITE'

   #define E1_DIR_WRITE(STATE) WRITE(E1_DIR_PIN,STATE)

                               ^

sketch\stepper_indirection.h:442:108: note: in expansion of macro 'E1_DIR_WRITE'

     #define REV_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                                                                                                            ^

sketch\stepper_indirection.h:442:121: note: in expansion of macro 'INVERT_E1_DIR'

     #define REV_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                                                                                                                         ^

sketch\stepper.cpp:309:7: note: in expansion of macro 'REV_E_DIR'

       REV_E_DIR();

       ^

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\stepper.cpp:47:

Configuration.h:759: error: 'True' was not declared in this scope

 #define INVERT_E1_DIR True

                       ^

sketch\fastio.h:49:36: note: in definition of macro '_WRITE_C'

 #define _WRITE_C(IO, v)   do { if (v) { \

                                    ^

sketch\fastio.h:74:21: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V) _WRITE(IO,V)

                     ^

sketch\stepper_indirection.h:315:31: note: in expansion of macro 'WRITE'

   #define E1_DIR_WRITE(STATE) WRITE(E1_DIR_PIN,STATE)

                               ^

sketch\stepper_indirection.h:441:110: note: in expansion of macro 'E1_DIR_WRITE'

     #define NORM_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(!INVERT_E0_DIR); } else { E1_DIR_WRITE(!INVERT_E1_DIR); } }

                                                                                                              ^

sketch\stepper_indirection.h:441:124: note: in expansion of macro 'INVERT_E1_DIR'

     #define NORM_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(!INVERT_E0_DIR); } else { E1_DIR_WRITE(!INVERT_E1_DIR); } }

                                                                                                                            ^

sketch\stepper.cpp:313:7: note: in expansion of macro 'NORM_E_DIR'

       NORM_E_DIR();

       ^

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\stepper.cpp:47:

Configuration.h:759: error: 'True' was not declared in this scope

 #define INVERT_E1_DIR True

                       ^

sketch\fastio.h:47:36: note: in definition of macro '_WRITE_NC'

 #define _WRITE_NC(IO, v)  do { if (v) {DIO ##  IO ## _WPORT |= _BV(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~_BV(DIO ## IO ## _PIN); }; } while (0)

                                    ^

sketch\fastio.h:74:21: note: in expansion of macro '_WRITE'

 #define WRITE(IO,V) _WRITE(IO,V)

                     ^

sketch\stepper_indirection.h:315:31: note: in expansion of macro 'WRITE'

   #define E1_DIR_WRITE(STATE) WRITE(E1_DIR_PIN,STATE)

                               ^

sketch\stepper_indirection.h:441:110: note: in expansion of macro 'E1_DIR_WRITE'

     #define NORM_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(!INVERT_E0_DIR); } else { E1_DIR_WRITE(!INVERT_E1_DIR); } }

                                                                                                              ^

sketch\stepper_indirection.h:441:124: note: in expansion of macro 'INVERT_E1_DIR'

     #define NORM_E_DIR() { if (current_block->active_extruder == 0) { E0_DIR_WRITE(!INVERT_E0_DIR); } else { E1_DIR_WRITE(!INVERT_E1_DIR); } }

                                                                                                                            ^

sketch\stepper.cpp:313:7: note: in expansion of macro 'NORM_E_DIR'

       NORM_E_DIR();

       ^

exit status 1
'True' was not declared in this scope

Tannoo, 2017-04-30:
Strange, I don't have a dual extruder yet.
I set the following to see about getting the same error:
`#define MOTHERBOARD BOARD_RAMPS_14_EEB`
`#define EXTRUDERS 2`
`#define TEMP_SENSOR_0 1`
`#define TEMP_SENSOR_1 1`
`#define INVERT_E0_DIR false`
`#define INVERT_E1_DIR true`

and....it compiles just fine for me.

```
Sketch uses 218282 bytes (85%) of program storage space. Maximum is 253952 bytes.
Global variables use 6114 bytes (74%) of dynamic memory, leaving 2078 bytes for local variables. Maximum is 8192 bytes.
```

How about sending your whole config files this way?

Tannoo, 2017-04-30:
>#define INVERT_E1_DIR True

Wouldn't the `True` be a problem? Try `true` instead.

pmbroth, 2017-04-30:
YOu are correct - it did not like "True", but worked with "true"  Thank you!!!

pmbroth, 2017-04-30:
I'm so happy!!!  :-)  I was going crazy...

closed 2017-04-30 by pmbroth