MarlinFirmware/Marlin #6503: last commands from SD stucked by wait_for_user

cormier8, 2017-04-30:
I'm using RCBugFix on a K8400 dual heads 3D printer, with SD card and LEDs effects enabled.

I observed that sometime, the LCD displays "Completed" and wait for abutton pressed, but the last commands from SD card are not executed.

As the last commands are for turn off heaters and park the head, this may damage the print.

If I suppress the while (wait_for_user) loop below, all commands from SD car are executed.

So I think that when card's eof is encountered and Marlin waits for user to press button, some SD commands are still in the command buffer.

The concerned code is in Marlin_main.cpp, in the get_sdcard_commands() procedure :

        if (card_eof) {
          SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
          card.printingHasFinished();
          #if ENABLED(PRINTER_EVENT_LEDS)
            LCD_MESSAGEPGM(MSG_INFO_COMPLETED_PRINTS);
            set_led_color(0, 255, 0); // Green
            #if HAS_RESUME_CONTINUE
              KEEPALIVE_STATE(PAUSED_FOR_USER);
              wait_for_user = true;
              while (wait_for_user) idle();    <-------- loop wich may prevent from executing last commands
              KEEPALIVE_STATE(IN_HANDLER);
            #else
              safe_delay(1000);
            #endif
            set_led_color(0, 0, 0);   // OFF
          #endif
          card.checkautostart(true);
        }

Would it be possible to put Marlin in wait mode only after having check that the command buffer is empty ?

Tannoo, 2017-04-30:
Click the encoder and it will continue.
I had implemented a timeout for that also if the user didn't click. But, that didn't make the cut.

cormier8, 2017-04-30:
Hi Tannoo,

I observed that if I click the encoder the print continues.

The problem is that the user will not be near the printer when the card eof is reached.

In my case, I came back to the printer several hours after after the eof was hit and as the last commands were not executed (turn heater off and park the head), the print was burn.

So I maintain that Marlin should'nt go to wait mode before the last command is executed.

thinkyhead, 2017-05-04:
Thanks for pointing out this issue. I'm not sure what the proper solution is. See if this works:
```cpp
    if (card_eof) {
      SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
      card.printingHasFinished();
      #if ENABLED(PRINTER_EVENT_LEDS)
        LCD_MESSAGEPGM(MSG_INFO_COMPLETED_PRINTS);
        set_led_color(0, 255, 0); // Green
        #if HAS_RESUME_CONTINUE
          enqueue_and_echo_commands_P("M0");
        #else
          safe_delay(1000);
        #endif
        set_led_color(0, 0, 0);   // OFF
      #endif
      card.checkautostart(true);
    }
```

thinkyhead, 2017-05-04:
Thanks for pointing out this issue. I'm not sure what the proper solution is. See if this works:
```cpp
    if (card_eof) {
      SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
      card.printingHasFinished();
      #if ENABLED(PRINTER_EVENT_LEDS)
        LCD_MESSAGEPGM(MSG_INFO_COMPLETED_PRINTS);
        set_led_color(0, 255, 0); // Green
        #if HAS_RESUME_CONTINUE
          enqueue_and_echo_commands_P(PSTR("M0"));
        #else
          safe_delay(1000);
        #endif
        set_led_color(0, 0, 0);   // OFF
      #endif
      card.checkautostart(true);
    }
```

cormier8, 2017-05-06:
Works fine, thank's Thinkyhead.

closed 2017-05-06 by cormier8