MarlinFirmware/Marlin #6504: NOT AN ISSUE:  M119 code to return Filament Monitor status

al-jimenez, 2017-04-30:
I am new to this....  I saw that you are only suppose to submit bug fixes but could not find where to submit this....  a tiny bit of code.  Forgive me and let me know where/how I should do this in the right place.   This is tiny addition to the Marlin RC8 code that I made.   But someone told me I should submit it.  So here it is...

M119 GCode will return the status of each end-stop.  So I decided to add some code to return the status of the filament monitor switch as well.   The following code works and was verified.   The below code can be useful in troubleshooting any connection issues with the filament sensor switch.    

In "endstops.cpp" in function "void Endstops::M119()"  I added this code:
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
    SERIAL_PROTOCOLPGM(MSG_FIL_SENS);
    SERIAL_PROTOCOLLN( ( (READ(FIL_RUNOUT_PIN )^FIL_RUNOUT_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
 #endif

In "language.h", added:  #define MSG_FIL_SENS "Filament Sensor: "

Thats it.   Pls let me know if I should post this somewhere else.   I hope this is helpful.

Al Jimenez

Tannoo, 2017-04-30:
In RCBugFix, it's already there.
```
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    SERIAL_PROTOCOLPGM(MSG_FILAMENT_RUNOUT_SENSOR);
    SERIAL_PROTOCOLLN(((READ(FIL_RUNOUT_PIN)^FIL_RUNOUT_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
```

Tannoo, 2017-04-30:
```
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: open
filament: open
ok
```

al-jimenez, 2017-04-30:
Glad to hear it.

psavva, 2017-05-01:
@al-jimenez I think you can close this ticket if all is good :)

closed 2017-05-01 by al-jimenez