MarlinFirmware/Marlin #6510: UBL getting started guide

Bob-the-Kuhn, 2017-04-30:
Here's my first pass at a getting started guide for UBL.

**FEEDBACK IS ENCOURAGED**

Details on the G26 & G29 commands will also be added to the wiki at the same time.

[UBL startup guide.zip](https://github.com/MarlinFirmware/Marlin/files/966986/UBL.startup.guide.zip)

thinkyhead, 2017-04-30:
> Finer X Y resolution…

Bilinear bed leveling also allows up to a 16x16 grid

> User is able to fill in…

Mesh Bed Leveling includes this feature. However, it doesn't support Delta, thus its feature for allowing you to custom-set points is kind of superfluous.

Bilinear bed leveling is based on the delta leveling that was already in Marlin. So, while it automatically extrapolates points which can't be reached, and does a pretty good job, it doesn't yet include an option to set a Z value directly.

> A LCD display…

In English we typically use "an" in front of letters which are pronounced with a starting vowel. So "An LCD display," is better.

> The following command sequence is used…

It might be too early in the article to include those details, which users will simply skip over and/or forget. You may want to build up to this, or do a section on each command.

Semicolons are probably not the prettiest or most appropriate dividers on these lines. Either a colon, whitespace with consistent indentation, or an em-dash would be better.

---

Generally, it would be better to compose this in Markdown, using an editor that does Markdown preview (such as Atom), since that is the format it will be in. Formatting after-the-fact is much more difficult.

Roxy-3D, 2017-04-30:
I liked what Bob did!   I think the semi-colons are there so if the user sends them to Marlin they are a comment and don't affect anything.   I kind of like having the quick start list right at the front.  It lets people  start quickly!    I made a few changes to the list:
```cpp
The following command sequence can be used:  
M502 ;  only needed if EEPROM hasn’t been initialized
M500 ; only needed if EEPROM hasn’t been initialized
M501 ; only needed if EEPROM hasn’t been initialized
M190 S65 ; not required but having the printer at temperature helps accuracy
M104 S210; not required but having the printer at temperature helps accuracy
G28
G29 P1   ; automated probing of the bed will do G28 if not already done
G29 P3 O ; repeat until all mesh points are filled in
G29 O ; view the Z compensation values
G29 A ; Activate the UBL System
G29 S 1; save the values to EEPROM
M500 ; save current setup with UBL active at power up
G26 C P O3.0 ; produce mesh validation pattern with primed nozzle
G29 P4 O ; move nozzle to ‘bad’ areas and fine tune the values if needed
Repeat G26 and G29 P4 O as needed
G29 S 1 ; save the values to EEPROM
```

playlet, 2017-05-01:
Is it possible to use UBL entirely via LCD? I don't want to use the computer to do all these steps

Roxy-3D, 2017-05-01:
> Is it possible to use UBL entirely via LCD? I don't want to use the computer to do all these steps

Not yet.   But I'm sure somebody will add a UBL menu to the LCD panel soon enough.   But the thing is...   If you connect up a computer long enough to get a mesh fine tuned, you don't need it after that.

pmbroth, 2017-05-02:
Hello,  My build plate is 16" by 28.5", will I be able to use this?

Roxy-3D, 2017-05-02:
Yes.   The system is designed to give even large beds like that good adheasion across the entire bed.

Tannoo, 2017-05-02:
>But Tannoo is working through the issues to do that!

Crap... Now, the pressure is on.

Roxy-3D, 2017-05-02:
Na...    He can use his laptop to talk to the printer until the LCD Code is all buttoned up and working.

Tannoo, 2017-05-02:
Why is `M190 S50` working, but enqueing M600?

```
echo:PID settings:
echo:  M301 P15.25 I0.93 D79.87
echo:  M304 P358.06 I49.23 D651.00
echo:Z-Probe Offset (mm):
echo:  M851 Z-4.05
echo:DEBUG:ECHO
ok
echo:SD card ok
echo:M190 S50
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:0 W:?
echo:enqueueing "M600"
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:127 W:?
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:127 W:?
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:127 W:?
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:127 W:?
 T:31.2 /0.0 B:40.9 /50.0 @:0 B@:127 W:?
 T:31.2 /0.0 B:41.1 /50.0 @:0 B@:127 W:?
...
```

Tannoo, 2017-05-03:
I have now submitted #6566.

Bob-the-Kuhn, 2017-05-06:
@Roxy-3D, @thinkyhead  - I took a shot on just dropping the current G29 header into the wiki.  The current style just isn't supported by Markdown.  Below is the best I could come up with.

[image]

Roxy-3D, 2017-05-06:
I think for the Wiki...   One line is all that is going to be tolerated.  Especially with the number of parameters.   

For now...  I'm debating the same thing on G26.  I think I'll condense each option to 1 line.  But so far... I've been telling them to go look at the source file:  http://reprap.org/wiki/G-code#G26:_Mesh_Validation_Pattern

Roxy-3D, 2017-05-07:
Bob...  I haven't forgotten about this...   How about we move it to the documentation repository and we just edit it over there.   In the mean time, it will be a work in place but it can be helping people???

lrpirlet, 2017-05-07:
@Bob-the-Kuhn, thanks for this paper, in a few pages it gives a very nice overview of the UBL..

I am missing the description of some of the G26 parameters put forward by @Roxy-3D, namely  G26 C...

I feel somewhat surprised to observe that m48 ends with **bed leveling on** while G28 ends with **bed leveling off**... This is the invert of previous MBL where M48 would disable MBL while G28 would keep it enabled... @Roxy-3D: Is that on purpose???

Roxy-3D, 2017-05-07:
I think it is just a difference of opinion.    Remember...  We don't have a well thought out GCode manual for 3D-Printers.   Things are evolving fast and different perspectives and visions need to be worked through.

I'll try to get the G28 issue changed today.

naptalene, 2017-05-09:
Hi,

I just jumped from Marlin 1.0.0 to 1.1.0.
As a noob I got the printer running perfectly except for the bed levelling (I chose the UBL option).
After searching I came across the guide posted above which explains the setup process and main commands very succinctly. I feel confident that after reading it I can get it working and at least be able to ask informed questions if I can't.

May I suggest that you post it on the main Marlin.org site with the bed leveling documentation if you are happy with it from a technical point of view.
I'm sure there will be many others who will benefit from it.

coolest:)

Bob-the-Kuhn, 2017-05-09:
Scott just re-built the wiki so the guide is now visible to the world!!!

http://marlinfw.org/docs/features/unified_bed_leveling.html

naptalene, 2017-05-10:
My timing was impeccable lol.
Thanks for the guide btw

Bob-the-Kuhn, 2017-05-10:
@thinkyhead - thanks for doing the edits - looks much better.

Bob-the-Kuhn, 2017-05-11:
@Roxy-3D - here's another pass at moving the G29 details into the wiki. 

I moved the phase commands to the start because of limitations within markdown.  

Ignore the blank table elements.  They came about when I pasted the preview into word.

I need to start looking at the text & bring it up to date.

[G29 md 1.zip](https://github.com/MarlinFirmware/Marlin/files/991555/G29.md.1.zip)

Roxy-3D, 2017-05-11:
Thank You...   I'll leave this page open so I can review it in the morning.   (You don't want me to review it right now.   I was eating cheese and cracker and drinking a bottle of wine while watching a movie)

Roxy-3D, 2017-05-11:
Those table elements are being difficult...   But... I'll get it reviewed...   I'll turn revision tracking on so you can see what changed and make a decision about what changes to keep.

Bob-the-Kuhn, 2017-05-11:
Just update the headers in G26 & G29 files and I'll copy them over.

I just wanted you to see the table format.

Roxy-3D, 2017-05-11:
Oh!   I understand.   One problem with the table format is it keeps the text from being compact.   

Does mark down not have any good ways to deal with and display this kind of information?

Bob-the-Kuhn, 2017-05-11:
It doesn't have a good method.

Take a look around the wiki & see if you find something better.

The only other alternative is to ask Scott to do some CSS magic & define something specifically for this application.

Roxy-3D, 2017-05-12:
OK...  I've finally been able to start the review and edit process.   Probably I'll just do 'Direct Commits' because the formatting takes a number of iterations to get it to look right.   I'll just have an edit window open and a display window.

Bob-the-Kuhn, 2017-05-12:
?? Are you editing the cpp file or the docx or ??

Just bring the cpp file up to date & I'll move it into the wiki.

Roxy-3D, 2017-05-12:
The https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md file.   It needs some changes.

Roxy-3D, 2017-05-14:
@Bob-the-Kuhn   Which one is the master copy?   I can edit https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

But it looks like http://marlinfw.org/docs/features/unified_bed_leveling.html is more complete and mature?   The problem is, I can't edit this one.

Bob-the-Kuhn, 2017-05-14:
The .MD file is the source.  The HTML file is generated from the MD file.

Roxy-3D, 2017-05-14:
OK...  That explains things.   Both locations will have identical information?   

Which domain hold the .MD file that should be edited?

Bob-the-Kuhn, 2017-05-14:
When I download the ZIP & unzip it the file is at 
MarlinDocumentation-master\_features\unified_bed_leveling.md

Here's what the current file looks like:
[unified_bed_leveling.zip](https://github.com/MarlinFirmware/Marlin/files/999779/unified_bed_leveling.zip)

Roxy-3D, 2017-05-14:
OK...   I get it...  But then what is the purpose of http://marlinfw.org/docs/features/unified_bed_leveling.html

Bob-the-Kuhn, 2017-05-14:
They're a byproduct of the build/generate process used to test/verify the changes.  They could all disappear from the repository with no adverse consequences.

Roxy-3D, 2017-05-15:
So...   If I'm reviewing and editing documents...  I  edit it here: https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

Roxy-3D, 2017-05-22:
I have not forgotten that I'm supposed to do this.   Today I really tried to get started.  But I just wasn't in the mood to do documentation.  Documentation is very hard for me.  I don't write well.   So instead, I cleaned up a bunch of stuff and added some nice user features instead.  https://github.com/MarlinFirmware/Marlin/pull/6822

Plus...  I had to get my printer back together again, test it, calibrate the new probe, create new mesh data, etc.    

There is a good chance I'll be in the mood tomorrow to start editing....   At least, we can cross our fingers and hope.

Bob-the-Kuhn, 2017-05-22:
You could print it out, mark it up, scan it & then post it here.  I'll be happy to update the wiki.

The G29 header still says that G29 P4 defaults to editing every point in the mesh.  I'd rather you go through the G26 & G29 headers first and then mark up the wiki.

Actually, you could mark both of them up & I'll take care of making them pretty.

vido89, 2017-09-26:
Thank you for sharing/writing this article, Im waiting for delta UBL support

Roxy-3D, 2017-09-26:
> Thank you for sharing/writing this article, Im waiting for delta UBL support

It is there.   The code supports it.   It just isn't in the guide yet.

closed 2017-10-17 by Bob-the-Kuhn