MarlinFirmware/Marlin #6512: BLTOUCH menu items missing macro's

Hertog99, 2017-04-30:
With the latest RCBugFix (671a44b8aac2d38123ef95da980c6f2146f1f1be I believe) there seem to be missing a bunch of definitions for the display menu's that get enabled when enabling BLTOUCH?

```
In file included from sketch/ultralcd.cpp:61:0:
sketch/ultralcd.cpp: In function 'void bltouch_menu()':
ultralcd.cpp:737: error: 'MSG_BLTOUCH_DEPLOY' was not declared in this scope
       MENU_ITEM(gcode, MSG_BLTOUCH_DEPLOY, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_DEPLOY)));
                        ^
sketch/ultralcd_impl_DOGM.h:750:114: note: in definition of macro 'lcd_implementation_drawmenu_gcode'
   #define lcd_implementation_drawmenu_gcode(sel, row, pstr, gcode) lcd_implementation_drawmenu_generic(sel, row, pstr, '>', ' ')
                                                                                                                  ^
sketch/ultralcd.cpp:253:7: note: in expansion of macro '_MENU_ITEM_PART_1'
       _MENU_ITEM_PART_1(TYPE, LABEL, ## __VA_ARGS__); \
       ^
sketch/ultralcd.cpp:737:7: note: in expansion of macro 'MENU_ITEM'
       MENU_ITEM(gcode, MSG_BLTOUCH_DEPLOY, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_DEPLOY)));
       ^
ultralcd.cpp:738: error: 'MSG_BLTOUCH_STOW' was not declared in this scope
       MENU_ITEM(gcode, MSG_BLTOUCH_STOW, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_STOW)));
                        ^
sketch/ultralcd_impl_DOGM.h:750:114: note: in definition of macro 'lcd_implementation_drawmenu_gcode'
   #define lcd_implementation_drawmenu_gcode(sel, row, pstr, gcode) lcd_implementation_drawmenu_generic(sel, row, pstr, '>', ' ')
                                                                                                                  ^
sketch/ultralcd.cpp:253:7: note: in expansion of macro '_MENU_ITEM_PART_1'
       _MENU_ITEM_PART_1(TYPE, LABEL, ## __VA_ARGS__); \
       ^
sketch/ultralcd.cpp:738:7: note: in expansion of macro 'MENU_ITEM'
       MENU_ITEM(gcode, MSG_BLTOUCH_STOW, PSTR("M280 P" STRINGIFY(Z_ENDSTOP_SERVO_NR) " S" STRINGIFY(BLTOUCH_STOW)));
       ^
sketch/ultralcd.cpp: In function 'void lcd_prepare_menu()':
ultralcd.cpp:1776: error: 'endstops' was not declared in this scope
       if (!endstops.z_probe_enabled && TEST_BLTOUCH())
            ^
exit status 1
'MSG_BLTOUCH_DEPLOY' was not declared in this scope
```
Could be that I am falling right in the middle of some frantic merging going on, but just in case, I'd thought I notify you of this happening.

This is with a fresh pull, only things I changed was enabling BLTOUCH and enable DOGLCDand REPRAP_DISCOUNT_FILL_GRAPHIC_SMART_CONTROLLER

Roxy-3D, 2017-04-30:
> Could be that I am falling right in the middle of some frantic merging going on, but just in case, I'd thought I notify you of this happening.

In theory... No.  Travis checks all these simple type of errors for a huge number of permutations of printers.    But some BL-Touch stuff snuck into the release in the last few days and it is very possible something got broke....

Thanks for the Heads Up!

Hertog99, 2017-04-30:
The following (trivial?, of course they never are ;)) adds the missing macro's, and fixes the compile:
```
diff --git a/Marlin/language_en.h b/Marlin/language_en.h
index 35a0cbb8..80f17ea5 100644
--- a/Marlin/language_en.h
+++ b/Marlin/language_en.h
@@ -417,6 +417,12 @@
 #ifndef MSG_BLTOUCH_RESET
   #define MSG_BLTOUCH_RESET                   _UxGT("Reset BLTouch")
 #endif
+#ifndef MSG_BLTOUCH_DEPLOY
+  #define MSG_BLTOUCH_DEPLOY                  _UxGT("Deploy BLTouch")
+#endif
+#ifndef MSG_BLTOUCH_STOW
+  #define MSG_BLTOUCH_STOW                    _UxGT("Stow BLTouch")
+#endif
 #ifndef MSG_HOME
   #define MSG_HOME                            _UxGT("Home") // Used as MSG_HOME " " MSG_X MSG_Y MSG_Z " " MSG_FIRST
 #endif
diff --git a/Marlin/ultralcd.cpp b/Marlin/ultralcd.cpp
index 16952ac7..feca0ea4 100755
--- a/Marlin/ultralcd.cpp
+++ b/Marlin/ultralcd.cpp
@@ -720,6 +720,7 @@ void kill_screen(const char* lcd_msg) {
   #endif // MENU_ITEM_CASE_LIGHT
 
   #if ENABLED(BLTOUCH)
+    #include "endstops.h"
 
     /**
      *
```

Not gitlab-savvy enough yet to make a proper pull request out of it.

bgort, 2017-05-01:
I'm seeing the same errors.  I'll test out and then turn your patch into a pull request if all is working.  Thanks very much.

bgort, 2017-05-01:
Ugh - these two messages need to be translated into how many languages?  I don't know but a few of these....

Roxy-3D, 2017-05-01:
> Ugh - these two messages need to be translated into how many languages? I don't know but a few of these....

I believe the correct proceedure is to just put the message in English into the language.h file...   And if there is not a local language override...  That is what gets displayed.

bgort, 2017-05-01:
Yep.  I realized that when I added the English and then looked at the french file.  Thanks!

closed 2017-05-03 by thinkyhead