MarlinFirmware/Marlin #6518: ultralcd.cpp missing include for BLTouch support

mperdue, 2017-05-01:
I got an error when compiling the latest rcbugfix telling me that endstops was not defined. The include for endstops.h seems to have been removed.

I added the following @ line 37 in ultralcd.cpp to correct the issue.

#if ENABLED(BLTOUCH)
  #include "endstops.h"
#endif

Also, BLTouch strings for stow and deploy need to be added to the language files.

bgort, 2017-05-01:
Just fixed these in #6517.

Even with them fixed, Marlin freezes at startup after deploying the BLTouch once with a 'KILLED.  Printer Halted.'  Can you let me know if you're seeing that also?

Roxy-3D, 2017-05-01:
> Marlin freezes at startup after deploying the BLTouch once with a 'KILLED. 

I added some code to turn the heaters off while the BL-Touch was probing because it was causing accuracy  problems with the HAL effect sensor.    That code got altered in: 
 https://github.com/MarlinFirmware/Marlin/pull/6491  To do a Kill.   My guess....  the logic got over simplified.

thinkyhead, 2017-05-01:
> That code got altered in: #6491 To do a Kill.

Sorry, that was supposed to be temporary for testing. I will finish and patch that.

> accuracy problems with the HAL effect sensor

"Hall effect" sensor. Named after Edwin Hall.

thinkyhead, 2017-05-01:
Should be resolved now. Please confirm.

bgort, 2017-05-01:
Thanks.  Will confirm it's working shortly.

bgort, 2017-05-01:
984c210dec4fdc03d9f5df77ad897cef8c64f692 still has the problem.

Commenting out BLTOUCH_HEATERS_OFF fixes it.  I'll go looking for the issue and will see about fixing it after I get some Monday-morning things out of the way.

thinkyhead, 2017-05-03:
Should be all fixed now!

PR #6519 (2017-05-01): Various fixes for compiler warnings
Resolve conflicts, rebase, clean up, and extend #6517

- add BLTouch-related messages in english and (rusty) french;
- add missing `endstops.h` in `ultralcd.cpp`
- fix misc. compiler warnings
- fix `lsf_reset` - `ZERO` macro can't be used with `struct` objects, only arrays

Also:
- Fix `set_heaters_for_bltouch`  to resolve #6518
- Remove executable bit from UBL and other file permissions
closed 2017-05-03 by thinkyhead