MarlinFirmware/Marlin #6521: inline keyword has no effect in code size

trivalik, 2017-05-01:
Any idea how this should be? I removed several "inline" in the Marlin_main.cpp. But not effect.

JustAnother1, 2017-05-01:
inline tells the compiler to not create a  callable code peace for that, but to put the code everywhere this function is called. So if a inline function is just called once then the code size will be about the same with or without the inline. If the inlined function is called from thousands of places then the code size will grow with inline enabled. inline is usually used to speed up the code. instead of calling a function the functions code is executed. This saves a few instructions due to the skipped function call.

Does this answer your question?

thinkyhead, 2017-05-01:
In fact, `inline` only _suggests_ to the compiler that it can put the function inline. The compiler may or may not do so. And if you leave off `inline` the compiler may still decide to inline the function anyway.

closed 2017-05-01 by thinkyhead

trivalik, 2017-05-01:
Sure, you are alright with that. Sorry.