MarlinFirmware/Marlin #6522: MBL and EEPROM Settings

james94jeans2, 2017-05-01:
When using Mesh Bed Leveling and saving the settings to EEPROM not the full mbl.status is saved, but rather only the has_mesh bit.

This results in the printer forgetting about mbl beeing turned on.

Solutions could be to save the full status integer (EEPROM waste?) or storing the mbl.is_active bit instead of the mbl.has_mesh bit and when loading assuming that there is a mesh when mbl is active.

Another thing concerning MBL: 

Maybe there should be a hint, that MBL and manual_probing should not be enabled at the same time.

Tannoo, 2017-05-02:
See this PR: #6537.

thinkyhead, 2017-11-01:
Printer starts up in unknown state with leveling off. `G28` includes in its specification that it disables bed leveling. `M420 S1` should be used to enable leveling when desired. It's recommended to add this command to your slicer's Start G-code — in Cura this is stored as part of the printer profiles.

closed 2017-11-01 by thinkyhead