MarlinFirmware/Marlin #6525: DELTA_HEIGHT not declared in this scope

LeavOA, 2017-05-01:
Merging the latest RCBugFix into my modified code after a while, and this came up:

```
C:\Users\Leav\AppData\Local\Temp\arduino_build_798324\sketch\Marlin_main.cpp: In function 'void gcode_M665()':

Marlin_main.cpp:7635: error: 'DELTA_HEIGHT' was not declared in this scope

       home_offset[Z_AXIS] = code_value_linear_units() - DELTA_HEIGHT;

                                                         ^

Using library Wire at version 1.0 in folder: C:\Users\Leav\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.18\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidTWI2 (legacy)
exit status 1
'DELTA_HEIGHT' was not declared in this scope

```

Quite possibly due to an error on my part? but I can't find any reference to a `#define DELTA_HEIGHT` on the RCBugFix branch so this addition might have been forgotten?

LeavOA, 2017-05-01:
in the section:
```
//===========================================================================
//============================== Delta Settings =============================
//===========================================================================
```

added 
```
  //Delta total height
  #define DELTA_HEIGHT 470.0
```
and it compiled fine (haven't uploaded and tested yet).

Could the real issue be that the example configurations haven't been updated for RCBugFix yet?

thinkyhead, 2017-05-02:
`DELTA_HEIGHT` was added to the example configurations several days ago, so that's all set. I've patched the code so that if `DELTA_HEIGHT` is not provided then it will be auto-assigned.

closed 2017-05-02 by thinkyhead