MarlinFirmware/Marlin #6526: KILLED. PRINTER HALTED

NiklasOlofsson, 2017-05-01:
Hello,

I see many changes in RCBugFix the last day. I used the new config and transferred my settings to the new config structure. Stuff has changed places. :)

Now, the printer halts on power on, after the splash screen I get "KILLED. PRINTER HALTED, Please reset."
I can't find what I need to change to get it working again. I used my older config straight off with the latest checkout and still getting the same message. 

Log:

> Recv: echo: M851 Z-1.00
> Recv: Error:Printer halted. kill() called!
> Changing monitoring state from 'Operational' to 'Error: Printer halted. kill() called!\x0a'
> Changing monitoring state from 'Error: Printer halted. kill() called!\x0a' to 'Offline'

thinkyhead, 2017-05-01:
Please post your configuration files. You can either Zip them together or append `.txt` to the filenames, and then Github will allow you to drop them on the reply field.

NiklasOlofsson, 2017-05-01:
This is my old config that I used with older RCBugFix:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/968206/Marlin.zip)
Just dropping it in with the latest code kills the printer.

And this is my new configs:
[Marlin_new.zip](https://github.com/MarlinFirmware/Marlin/files/968209/Marlin_new.zip)
Same kill with these.

thinkyhead, 2017-05-01:
Try [this branch](https://github.com/thinkyhead/Marlin/tree/rc_cleanups_2) and let me know if it fixes the issue.

thinkyhead, 2017-05-01:
That seems to be it! Applied the patch to the main branch.

NiklasOlofsson, 2017-05-01:
Ah, great. Thanks! Will try when done printing. 👍

NiklasOlofsson, 2017-05-01:
And it works! Thanks.

NiklasOlofsson, 2017-05-01:
Spoke to soon. Now it does not heat the bed.
The display show target temp 60 but the temperature does not change and the result is halted printer, heating failed.

cbusillo, 2017-05-02:
I have the same issue.  RCBugFix is up to date.  Turning off BLTOUCH_HEATERS_OFF fixes it for now.

NiklasOlofsson, 2017-05-02:
I have BLTOUCH_HEATERS_OFF disabled but the bed or hotend does not heat up. Will try latest code now.

NiklasOlofsson, 2017-05-02:
Nope, no heating with BLTOUCH_HEATERS_OFF disabled. Can set target temp bud the bed or hotend does not heat up.

bgort, 2017-05-02:
Anything else unusual going on, @NiklasOlofsson?  It's -just- that the heaters aren't heating?  No more 'KILLED' message?

NiklasOlofsson, 2017-05-02:
Not the first problem, got killed at boot time. That's gone.
Now you get "heating failed" after some time. Target temps change but no power is going to the hotend or bed.

bgort, 2017-05-02:
Your pins_GT2560.h are identical between old-working and new-notworking?  

What was the commit id of the last version that worked for you?

NiklasOlofsson, 2017-05-02:
Yes. No changes to the pins.

I don't know commit really. Checked out 2017-04-30 20:23 (looking at the file create date)

Not alone :)
https://github.com/MarlinFirmware/Marlin/issues/6544

bgort, 2017-05-02:
4/30 20:23 is working or non-working?

NiklasOlofsson, 2017-05-02:
Working.

bgort, 2017-05-02:
If you can, in the directory where you checked out the working version, do a 'git log' and the commit id will be the first line output.

NiklasOlofsson, 2017-05-02:
Sorry, I can't. Downloaded as zip.

bgort, 2017-05-02:
If you still have the zipfile, the commit is in the attached notes, like here:

[image]

The '99b1...' is the last commit included in the archive.

Sorry for the hassle -- just trying to get to what works for you so I can compare it with what doesn't work.

NiklasOlofsson, 2017-05-02:
I don't have the zip file. Where does that note come from?

bgort, 2017-05-02:
It's embedded in the archive as comments.  It's not in a separate file, as far as I know.

NiklasOlofsson, 2017-05-02:
This is the re-zipped dir containing my working version
[Marlin-RCBugFix-old.zip](https://github.com/MarlinFirmware/Marlin/files/971071/Marlin-RCBugFix-old.zip)

bgort, 2017-05-02:
Your working version's pins.h contains:
```
#elif MB(GT2560)
  #include "pins_GT2560.h"
```

While the latest doesn't.

So unless you're adding that, your pins aren't defined at all.  I assume that's the issue.

NiklasOlofsson, 2017-05-02:
What last? I add it myself and it's there in my latest checkout.

NiklasOlofsson, 2017-05-02:
This is the same problem?
https://github.com/MarlinFirmware/Marlin/issues/6544

bgort, 2017-05-02:
I was comparing to the latest RCBugFix.  I understand now -- you're adding your pins_GT2560.h.  Didn't realize that.

bgort, 2017-05-02:
Send the .zip of the non-working version, if you don't mind.

NiklasOlofsson, 2017-05-02:
Yep. I have the "About printer" menu activated and I did not like it when it said "Ultimaker" in "Board Info". ;)

I have posted the non working version ^^

bgort, 2017-05-02:
I mean the whole directory - not just the configs.

bgort, 2017-05-02:
Ahh -- you edited it and I hadn't refreshed.  Got it now.

NiklasOlofsson, 2017-05-02:
Yeah, sorry. Posting before thinking. ;)

bgort, 2017-05-02:
No worries.

bgort, 2017-05-02:
I'm stumped right now.  In both working and non-working you have the new BLTOUCH_HEATERS_OFF disabled, and nothing else that touches the heaters has been changed between the versions.

I'll circle back to this in a bit.

NiklasOlofsson, 2017-05-02:
Well. Ok, I feel stupid, again.

Did not do M502 -> M500 as I always do when updating the FW...... 🥇
And yes, that did it! Now working again.

bgort, 2017-05-02:
Haha.  I almost suggested that, but thought 'surely he did....' and moved on .. glad you found it!

NiklasOlofsson, 2017-05-02:
Of course I did................................. ;)

Thanks for looking in to it. Wasting your time. Sorry. ;)

bgort, 2017-05-02:
No worries.  It happens!

closed 2017-05-02 by NiklasOlofsson

sathishcores, 2018-02-17:
Can anyone solve my prb

sathishcores, 2018-02-17:
Printer halted kill called

AnHardt, 2018-02-17:
Yes - you can.

You can usually find the reason in the serial output - right before the kill message.

SRIRAM134, 2019-02-22:
Hello, I m abhay...I have a DIY 3d printer (leadscrew driven)... Whenever I give print, the printer homes x,y and z axes and then after lots and lots of "echo:busy processing" messages, finally this comes "ERORR: Heating failed, Heating ID 0" and ERROR: Printer halted!.... 
I have J-head E3d hotend , no heated bed....
Please help.....

AnHardt, 2019-02-22:
@SRIRAM134 
There is no "J-head E3d hotend". It's eiter or - even when the Asian sellers try to make you believe the opposite. When they talk about "J-head E3d hotend" they mean a not specified random "hotend". They just include both of the types to be better found.

Having said that, it's most likely you don't know what thermistor type you have. (Most likely a 11 or 13)
At first make a reality check if the measured temperatures can be about right.
Is the heater getting warm at all, when you try to heat it.
Configs please.