MarlinFirmware/Marlin #6532: jogging mode similar to GRBL 1.1

w4sepp, 2017-05-01:
Jogging Mode : The new jogging commands are independent of the g-code parser, so that the parser state doesn't get altered and cause a potential crash if not restored properly. Documentation is included on how this works and how it can be used to control your machine via a joystick or rotary dial with a low-latency, satisfying response.

thinkyhead, 2017-05-04:
Looking forward to it. Currently we have `BABYSTEPPING` (which is independent of the GCode parser), in addition to LCD-based manual movement, which does update the cartesian current position. These features include support for a wheel as found on most controllers, or a keypad with forward-back buttons.

w4sepp, 2017-05-14:
Do you think it will be implemented?

thinkyhead, 2018-02-06:
How does this differ from `BABYSTEPPING`?

boelle, 2019-02-18:
i would dare and say that jog mode is the same as movement when not printing... so i have a hard time getting this to make sense

williamkapke, 2019-09-23:
The major difference is GRBL's special jog-cancel command (noted in an associated issue) which causes motion to immediately stop and clears the planner.

... the concept is good- however, for GRBL, it's pretty flawed and doesn't work very well. As they note in their docs- "it's not a complete solution": https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

The failure is that you need to stream jog commands at an interval and do your best to predict when the next command needs to be sent to keep the planner full- but not too full.

maxholgasson, 2024-03-03:
> extend the Marlin G-code parser for $J= jogging commands, but it's not high on the list.

That sounds ao awesome. Is there any progress in the last years of this issue?
I'm using the direct joystick support but only 3 are supported, it's not that smooth and it would be nice to do it with some kind of commands