MarlinFirmware/Marlin #6535: Make 3d printer using marlin to 2d printer

batata004, 2017-05-01:
Hi,

Is it possible to use some GCODE in the START SCRIPT of my Slicer to make the printer raise the Z exactly 1mm and wait I press a button in the LCD kit (the main button) to start print?

I succesfully made my reprap 3d printer become a 2d printer. It was very easy but I think I can make the printing better if before every print (after homing and auto bed leveling) the printer raise the nozzle 1mm and waits so I can put a pencil in a support and make the pencil touch the bed (instead of keeping the nozzle too close to the bed when 2d printing). So I would put the pencil and would press a button in the LCD so the printer starts drawing!

I already disabled the hotend and heated bed so they dont get warm while 2d printing cause it's completely useless.

 I ALSO would like to know if there is some easy way I can disable the extruder while 2d printing? The extruder is probably nos spinning because the hotend is not in a minimum temperature, but is there a way I can make sure the extruder gets disabled (GCODE) so even if my hotend gets hot the extruder wont spin?

stigjoergensen, 2017-05-01:
look at http://reprap.org/wiki/G-code

Specifically G0 / G1 and M0

teemuatlut, 2017-05-01:
> I ALSO would like to know if there is some easy way I can disable the extruder while 2d printing? The extruder is probably nos spinning because the hotend is not in a minimum temperature, but is [...]

Try [M111 S8](http://marlinfw.org/docs/gcode/M111.html) Debug Level: DRYRUN
In short it instructs Marlin to ignore any commands to heaters and extruder axis.

batata004, 2017-05-01:
@teemuatlut your suggestion is great but the docs says:

-> Don’t extrude, don’t save leveling data, etc

I need it to save leveling data cause right before printing I run an auto leveling instruction.

@stigjoergensen your idea is almost perfect. Maybe with a little bit more of your help (thanks!) we can make this perfect. I followed your orientation and my start code is:

G28
G29
G1 Z1
M0
G1 Z0.5

You see? The first G1 I use to raize Z to 1mm and wait. So I insert the pencil and make it touch the bed. After that I press a button and Z decreases to 0.5 and I WOULD LIKE IT TO START PRINT AT THIS HEIGHT! But unfortunatelly it goes down to 0.5 and when start printing it gets down back to 0 and the nozzle rips the paper.

How can I make the printer think the 0.5mm position is the zero? I tried G92 and it looks it does not work.

closed 2017-05-02 by batata004