MarlinFirmware/Marlin #6541: Anyone else having thermocouple issues with the RCBugFix?

Mind-FuLL, 2017-05-02:
Hello,
This is my first post so please go easy on me.  I don't have any official training in programming but I've been building my own machines for a few years now and I've learned a lot, especially from following what goes on here. I'm usually able to figure out what I need (or where I screwed up) by searching through here or some forums but I'm really stumped on this one.
The machine I’m working on uses a Azteeg X3 Pro with 3 extruders, two of which use thermocouples (-1).  This set up was already running on RC8.  I want to use the “DISTINCT_E_FACTORS” feature so yesterday I tried the latest bugfix after transferring all my settings and everything seems to work perfect except the thermocouples.  
At room temp they float around 11c instead of 24c.  The thermistors on the 3rd  extruder and bed are working perfectly.  The thermocouples track in the right direction when you turn the heaters on but something is obviously off.  When I reflash back to RC8 everything works again (except “DISTINCT_E_FACTORS” of course).  Back to Bugfix and 11c again.  Something’s up with the firmware but I can’t figure out what I could have done to cause this behavior.  
Has anyone else had thermocouple issues with the bugfix?  Is there some obvious thing I’m missing here that could be causing this?  I’ll post code my code if necessary. 

Nothing unusual about my setting other than I have applied [this code](https://github.com/MarlinFirmware/Marlin/issues/4220#issuecomment-238923131) to both the RC8 and bugfix, but I don’t think this would affect the thermocouples.

Regardless, thanks for all the work you guys put into this stuff!

Grogyan, 2017-05-02:
I use the Max31855, - 3, on RAMPS 1.4, not has any issues with the latest
RCBugFix

Mind-FuLL, 2017-05-02:
Got it. Forgot to change:
#define TEMP_SENSOR_AD595_GAIN   1.0
to
#define TEMP_SENSOR_AD595_GAIN   2.0

closed 2017-05-02 by Mind-FuLL