MarlinFirmware/Marlin #6543: marlin with lsee board

ahmadelwaly2016, 2017-05-02:
i'm use marlin firmware with lsee board but the printer doesn't work good what should i change in the firmware !! any help please 
[image]

Bob-the-Kuhn, 2017-05-02:
You'll need to be very specific as to what the problem is.

ahmadelwaly2016, 2017-05-02:
i connect the everything like in the attachment picture and download the marlin firmware when i connect the printer and open Printrun and connect " start
Printer is now online.
echo: External Reset
Marlin 1.0.0
echo: Last Updated: May  2 2017 16:28:46 | Author: (none, default config)
Compiled: May  2 2017
echo: Free Memory: 3822  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X78.82 Y78.82 Z400.00 E96.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X6000 Y6000 Z100 E7000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S2000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00
Error:0
: Extruder switched off. MINTEMP triggered !
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
echo:SD init fail
" 
this message appear for me and i can use the home position only to move the motor but i can't to move the motor one by one and the heater of bed and extruder didn't work but the connections right 
[image]

FHeilmann, 2017-05-02:
whatever the problem is your printer is not running the latest version of marlin, as apparent via the line that says "Marlin 1.0.0"

ahmadelwaly2016, 2017-05-02:
i update it now everything work okay but individual  ( manually ) but when i load a stl file the extruder heated but without outing any filament from the nozzle

psavva, 2017-05-03:
Have you configured Marin to use the correct steps/mm for your extruder?

At what temperature are you trying to extrude?
Is there any messages on the terminal output when trying to extrude?

It's probably a configuration of and sort

closed 2018-02-06 by thinkyhead