MarlinFirmware/Marlin #6544: Heating Failed Rigidboard V2

jaysonkelly, 2017-05-02:
I'm using the most recent version of RCBugFix and can't get my heaters to turn on. Any thoughts as to why? 

I am configured for Rigidboard V2 and the RIGIDBOT_PANEL controller display. 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/970824/Configuration.txt)

NiklasOlofsson, 2017-05-02:
Make sure that you do M502 followed by M500 when updating the FW......

jaysonkelly, 2017-05-02:
@NiklasOlofsson 
I've never had to do the M502 M500 sequence when updating firmware. I'll try that for sure when I get home. 
If it is so important is there any way to get the firmware to do it automatically on first boot?

NiklasOlofsson, 2017-05-02:
You could just disable eeprom in your configuration. I use eeprom to store my Z offset until I change it in my configuration.

NiklasOlofsson, 2017-05-02:
Before reporting any problems with the RCBugFix-branch we should always do the M502 - M500 and try again. I totally forgot to try that and I had the same no heat problem.

jaysonkelly, 2017-05-02:
There's a bunch of stuff I use eeprom to store. What causes the problem? Somone changing the structure of what gets stored to eeprom?

NiklasOlofsson, 2017-05-02:
I don't know what changed. @bgort or @thinkyhead probably knows. There have been some changes related to eeprom lately.

jaysonkelly, 2017-05-03:
I'm afraid resetting eeprom didn't fix the issue.

jaysonkelly, 2017-05-03:
What else could prevent heating?

jaysonkelly, 2017-05-03:
Disabling the SOFT_PWM_SCALE allows it to start heating.  Trying now to figure out why Soft Pwm Scale of 8 is causing failure to heat.

jaysonkelly, 2017-05-03:
Seems like it is using soft_pen_scale to shift a bit in a uint8_t variable in the isr() function. Shift it 8 times and you end up with zero. Recent modifications look like it is actually using this value differently. Changing the value to 7 fixes it. Might be a good idea to note that 7 is the max value for SOFT_PWM_SCALE in the comment above it in configuration.h

closed 2017-05-03 by jaysonkelly