MarlinFirmware/Marlin #6545: Make marlin HOME all axis and auto level

batata004, 2017-05-02:
Hi,

Many times I turn on my printer and after that I print lots of things sequentially. Before everyprint my printer home all axis and starts the auto leveling procedure.

So I ask you: if I remove the GCODE of my slicer that makes home all axis and BAL is it possible to make MARLIN everytime it boots up it does the home and BAL?

If my printer, everytime it was started could home all axis and BAL I wouldnt need to make it before every print. AND IF I FOR SOME REASON want to home all axis and BAL I can explicitly say so in the GCODE file my slicer produces.

It's not efficient having to manually click HOME AXIS, wait and do AUTO LEVELING using the LCD cause I lose at least 1 to 2 minutes. It would be nice this procedure would be possible to do right after the printer is turned on.

FHeilmann, 2017-05-02:
The problem with this is that some of us (myself included) need the printer to heat up before homing. 

Which leveling do you use? With some leveling features you can save the leveling matrix so you only have to home your axis before every print (which you can also speed up by increasing the homing feedrate in your config)

batata004, 2017-05-02:
Hi I am using a servo as a Bal. Isnt There a part in Marlin code that
allows me to run arbitráry code ?

batata004, 2017-05-02:
I mean isnt It possível tô Tell Marlin to run the Bal procedure as Soon as It is turnêd on?

batata004, 2017-05-02:
@FHeilmann how do I execute the utility script? I was not aware of it!

FHeilmann, 2017-05-02:
Do you print from SD or from a PC? 

Basically you create a .gcode file that only contains

```
G28
G29
```

and "print" this file as your first print of the day.

batata004, 2017-05-02:
GREAT! I USE SD!!! I didnt know that, amazing, thanks man! It would be nice to make this possible when the printer boots up but if it's not possible I am glad your suggestion works perfectly!

psavva, 2017-05-02:
@batata004 You also have the option of using UBL, which will save the bed levelling matrix into the EEPROM. 
This way, you can do the bed levelling one, and it will automatically restore the leveling matrix from the EEPROM once you turn on your printer.  

This is what I will be doing in the near future as well.

@Bob-the-Kuhn  has a nice guide explaining UBL in this Issue:
https://github.com/MarlinFirmware/Marlin/issues/6510

@Roxy-3D architected this awesome feature.

batata004, 2017-05-03:
UBL is the way to go :) Congrats guys you are doing a really amazing job with sooo many features in this firmware!

closed 2017-05-03 by batata004

thinkyhead, 2017-05-03:
You also have the option to use Mesh Bed Leveling, or a variety of Auto Bed Leveling procedures. All of these can save their results to EEPROM!