MarlinFirmware/Marlin #6546: SD Card Problem

spydermard, 2017-05-02:
hi!
I'm using RepRap Smart Controller for my printer.
[http://www.electrodragon.com/product/ramps-2004-lcd-smart-controller-reprap/](http://www.electrodragon.com/product/ramps-2004-lcd-smart-controller-reprap/)
but SD Card no work.
I'm using 2G SD Card, FAT32 format and Uncomment `#define SDSUPPORT`. but when I plug in SD Card it's not working.
what can I do to solve this problem?

Tannoo, 2017-05-02:
Try another card.

Cyper77, 2017-05-02:
format sd with that:
https://www.sdcard.org/downloads/formatter_4/index.html

spydermard, 2017-05-03:
Solved my problem. I use SD Card class 4 and it work easily. But why?

Tannoo, 2017-05-04:
1) Incompatability
2) Faulty card
3) Finiky card reader
4) Wrong class
5) Dirty contact pads on card
6) Navel lint in socket
7) Card covered in God knows what
8) Shorted pads on card
9) Damaged pad(s) on card
10) ...

spydermard, 2017-05-05:
tank's all.

closed 2017-05-05 by spydermard