MarlinFirmware/Marlin #6550: Compiling for Printrboard RevF fails in gcode_M900()

fiveangle, 2017-05-02:
Compiling RCBugFix @ ac69eca58604b3e16d1ce2d66f06a1f607747b0d I get:

```
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=136 -DARDUINO_ARCH_AVR -DARDUINO=10802 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\Marlin_main.cpp" -o "C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\Marlin_main.cpp.o"
In file included from C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\Marlin.h:42:0,

                 from C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\Marlin_main.cpp:223:

Marlin_main.cpp: In function 'void gcode_M900()':
serial.h:45: error: third operand to the conditional operator is of type 'void', but the second operand is neither a throw-expression nor of type 'void'
 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                                              ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:54:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\Marlin_main.cpp:8971:13: note: in expansion of macro 'SERIAL_ECHO'

     ratio ? SERIAL_ECHO(ratio) : SERIAL_ECHOPGM("Auto");

             ^

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
third operand to the conditional operator is of type 'void', but the second operand is neither a throw-expression nor of type 'void'
```

Compiling RCBugFix from approx Mar 20th has no issue.

Looks like it is likely related to commit eb2d073daef83a4a8203910613fd418fd320fbd as part of #6368

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/972031/Marlin-config.zip)

fiveangle, 2017-05-02:
if it helps track down the cause, disabling LIN_ADVANCE resulted in following change in compilation error:

```
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=136 -DARDUINO_ARCH_AVR -DARDUINO=10802 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp" -o "C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp.o"
In file included from C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp:23:0:

serial.h: In function 'void serialprintPGM(const char*)':
serial.h:35: error: 'customizedSerial' was not declared in this scope
   #define MYSERIAL customizedSerial

                    ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:96:42: note: in expansion of macro 'MYSERIAL'

   while (char ch = pgm_read_byte(str++)) MYSERIAL.write(ch);

                                          ^

serial.cpp: In function 'void serial_echopair_P(const char*, const char*)':
serial.h:35: error: 'customizedSerial' was not declared in this scope
   #define MYSERIAL customizedSerial

                    ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:45:46: note: in expansion of macro 'MYSERIAL'

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:54:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp:28:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, const char *v)   { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

serial.cpp: In function 'void serial_echopair_P(const char*, char)':
serial.h:35: error: 'customizedSerial' was not declared in this scope
   #define MYSERIAL customizedSerial

                    ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:41:25: note: in expansion of macro 'MYSERIAL'

 #define SERIAL_CHAR(x) (MYSERIAL.write(x))

                         ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp:29:81: note: in expansion of macro 'SERIAL_CHAR'

 void serial_echopair_P(const char* s_P, char v)          { serialprintPGM(s_P); SERIAL_CHAR(v); }

                                                                                 ^

serial.cpp: In function 'void serial_echopair_P(const char*, int)':
serial.h:35: error: 'customizedSerial' was not declared in this scope
   #define MYSERIAL customizedSerial

                    ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.h:45:46: note: in expansion of macro 'MYSERIAL'

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

[... 100 lines cut ...]
 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp:34:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, unsigned long v) { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

serial.cpp: In function 'void serial_spaces(uint8_t)':
serial.h:35: error: 'customizedSerial' was not declared in this scope
   #define MYSERIAL customizedSerial

                    ^

C:\Users\djohnson\AppData\Local\Temp\arduino_build_493716\sketch\serial.cpp:36:53: note: in expansion of macro 'MYSERIAL'

 void serial_spaces(uint8_t count) { while (count--) MYSERIAL.write(' '); }

                                                     ^

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
'customizedSerial' was not declared in this scope
```

Bob-the-Kuhn, 2017-05-03:
Dear Sir:   Please provide an error that's easier to fix.   This one is driving me nuts!

I was able to verify that the latest RCBugFix had almost identical error messages and that 18 Mar compiled OK.  This was with Arduino 1.6.12 & Teensy++ selected.  I don't know if my Teensyduino is up to date because I haven't been able to install it for a couple of months.

As best I can tell all the serial code leading up to requiring **customizedSerial** hasn't changed.

I'll look at it for another 30 minutes & then again in the morning.

Bob-the-Kuhn, 2017-05-03:
Before I forget ...  Shouldn't DAC_MOTOR_CURRENT_DEFAULT be enabled rather than DIGIPOT_MOTOR_CURRENT ?

thinkyhead, 2017-05-03:
I was never sure what the default should be —or if any of those should be enabled— so left it as the most common case. What's the effect?

thinkyhead, 2017-05-03:
Looking at the issue, the existence of `customizedSerial` depends on the conditions for defining it in `MarlinSerial.h`. Namely: `#ifndef USBCON`.

Bob-the-Kuhn, 2017-05-03:
DAC_MOTOR_CURRENT_DEFAULT is for the DAC on the Printrboard rev F.  The other is for a different DAC.

----

What I don't understand is why the older RCBugFix works but the new one doesn't.  The logic for deciding if `customizedSerial` is to be used or not is the same and the IDE extension is the same.

Off to attack it from a different angle.

fiveangle, 2017-05-03:
@Bob-the-Kuhn - yes, the DIGIPOT_MOTOR_CURRENT was an oversight in my line-by-line porting of my Configuration*.h to the new ones, but fixing that of course doesn't affect this issue.

I vaguely recall a similar USBCON issue for all AT90USB variant users sometime back and went poking around but didn't find any clues.  Out of desparation, reviewing #6400 

[Wasn't aware of the "details" tag to hide long test, I'll note that for future Scott - thanks !]

Bob-the-Kuhn, 2017-05-03:
@thinkyhead 

I'm using the Teensyduino IDE extension to compile the printr_board_revF.

This is definitely a pre-processor issue.  USBCON is not defined when **serial.h** is run but it is when **MarlinSerial.cpp** and **MarlinSerial.h** are run.  I inserted #error statements in the various USBON paths to generate this info.

What's REALLY weird is that **serial.h** depends on **MarlinSerial.cpp** to be run first or it'll fail on the non- AT90USB board.  

I feel like I'm missing something basic here about how the pre-processor does things and how many times it goes through the files before making the decision to include/exclude code.

It appears to me that USBCON is defined when **MarlinSerial.cpp** is run but later is not defined when **serial.h** is run.  Almost seems like a chicken & egg thing.

The only thing I noticed that could possibly be related is there is an "extra" compiler/linker flag in the latest RCBugFix Arduino compiler messages that is not present when the March RCBugFix is compiled.  Right after the -DLAYOUT_US_ENGLISH " section there's a **-I** .

`"C:\Program Files (x86)\Arduino 1.6.12\hardware\teensy/  ... -DLAYOUT_US_ENGLISH " -IC:\Program Files (x86)\Arduino 1.6.12\hardware\teensy\  ...`

Bob-the-Kuhn, 2017-05-03:
I forgot to mention one thing.

In order to get past the "ratio ? SERIAL_ECHO(ratio) : SERIAL_ECHOPGM("Auto");" error I commented it out in Malin_main.cpp.

thinkyhead, 2017-05-03:
> I wasn't aware of the "details" tag to hide long text

A great way to make long threads look short again!

thinkyhead, 2017-05-03:
> I commented it out in `Marlin_main.cpp`.

Patching now!

fiveangle, 2017-05-03:
@Bob-the-Kuhn - I found the same type of preproc time-space continuum rip when I dug into the USBCON definitions (using same method as you: just refining USBCON at various checkpoints).  Reviewing, I don't know how these other platforms work with #4820, as it appears #2433 should have been correct (except that the preproc doesn't seem to keep the USBCON definition from `iousb1286.h`->`iosubxx6_7.h` latched throughout the compile.

Bob-the-Kuhn, 2017-05-04:
As best I can tell USBCON comes & goes.  

I added the following code to the three places that use USBCON. xxxx is the name of the file.
```
  #ifdef USBCON
    #warning "xxxx  USBCON is defined"
  #endif

  #ifndef USBCON
    #warning "USBCON not defined (xxxx) "
  #endif
```

Basically it said that all three places get executed multiple times, USBCON can disappear and that eventually it gets caught in the gone state when it matters.

I tried to narrow down when/where it disappears by adding that code to a couple of files that appear to be compiling when it happens.  It disappears at "printcounter.cpp 1" and "qr_solve 1 " but is back in between.  

The attached is a copy of the compiler output from Arduino.  You'll see that all is well the first 15 or so times but then things get flakey.

I've tried putting #elif conditions just before the problem section is executed but no luck.

[printrboard 2 MAY warnings 1.txt](https://github.com/MarlinFirmware/Marlin/files/975443/printrboard.2.MAY.warnings.1.txt)

fiveangle, 2017-05-04:
Assuming that a gazzillion+1 programmers worldwide couldn't be wrong, I assumed this was a programatic change and not preproc flakieness so got a little more medievil in the search...

Ladies and gentlemen, I give you 2cbc7b4b73da8ae18dff7502cd520c551cfd4b7c.  No applause please, just throw money ;)

Now how to unwind it...

fiveangle, 2017-05-04:
[image]

Oh Scott, you are good ! ;)

Bob-the-Kuhn, 2017-05-04:
I've found a work around.  Don't know why it works.

The work around is to move serial.h & serial.cpp back into Marlin.h (like it was in March).

As best I can tell USBCON is not defined for a bit when the compiler is invoked.   When I put a test for USBCON at the beginning of the file and after every include the very first one always fails and none of the others ever fail.  If you put enough delay between the start of compilation and calling the troublesome code then the problem goes away.

fiveangle, 2017-05-04:
Including `Arduino.h` and `macros.h` in `serial.h` allows 1.1.0 to compile without any other changes.  Likewise including just `MarlinConfig.h` (which includes the above) also resolves, as does including just `Marlin.h` (which includes `MarlinConfig.h`).

I think @thinkyhead will have to review.

Bob-the-Kuhn, 2017-05-15:
The work around is on PR #6737

Bob-the-Kuhn, 2017-05-16:
Could you do me a big favor and see if the code in PR #6737 will compile with your configuration files?

It has the USBCON fix and the Teensyduino fix.  

It should be golden but ...  I spent 5 hours wrestling with the upload/update tools & I'm not sure everything survived the many edits involved.

fiveangle, 2017-05-27:
Closing since the includes added to serial.h resolved the original issue this was opened for.  Will open (or jump onto another) Teensy pin one if my testing this weekend presents more questions than answers :)

closed 2017-05-27 by fiveangle

Roxy-3D, 2017-06-07:
PrintRboard problems over at RepRap:  http://forums.reprap.org/read.php?415,772187

fiveangle, 2017-06-08:
Posted response.  That (this) issue was fixed a few weeks back.  They weren't using late enough bugfix to catch it.  Still need the other PRs for AT90USB merged, and then we won't be second class citizens any longer ;)

thinkyhead, 2017-06-10:
The PR unifying pin mappings was merged yesterday. I'm gratified by how simple (for the most part) that turned out to be!

Bob-the-Kuhn, 2017-06-14:
@fiveangle - need another favor.  Please see if the code in PR #7056 will compile and will set the DAC percents properly.

fiveangle, 2017-06-24:
Back from Warsaw... this was working fine for me 14 days ago without issue @Bob-the-Kuhn but I don't have LCD on right now (was testing UBL, which is too big to have LCD too).  Still need to review why it [doesn't report correct stepper amps tho](https://github.com/MarlinFirmware/Marlin/pull/7014#issuecomment-307539021)

I supposed I can disable UBL, enable LCD, and see if it dac get/set still works (LCD won't be connected).  Let me know if you still need that and I'll check it.

Bob-the-Kuhn, 2017-06-24:
Do what feels good to you.

I have a flat ribbon cable to connect the Printrboard Rev F to the MKS MINI12864.  If you want it I'll send it to you.  Otherwise I'll throw it in the trash.

FYI - I just created a RepRap Wiki page for the [MKS MINI12864](http://reprap.org/wiki/MKS_MINI_12864).  Seems rough to me.  I'd appreciate your feedback on it.

fiveangle, 2017-06-24:
I just recompiled with LCD support (LCD not connected) and it compiled without issue and was able to get/set through terminal:
```
Send: M909
Recv: echo:Stepper current values in % (Amps):
Recv: echo: X:70.00 (-17890.91) Y:70.00 (-17890.91) Z:50.00 (-23418.18) E:70.00 (-17890.91)
Recv: ok P15 B3
[...]
Send: M907 X50
Recv: ok P15 B3
[...]
Send: M909
Recv: echo:Stepper current values in % (Amps):
Recv: echo: X:50.00 (-23418.18) Y:70.00 (-17890.91) Z:50.00 (-23418.18) E:70.00 (-17890.91)
Recv: ok P15 B3
[...]
```
I hacked my LCD on the first time using single-pin jumpers (which is why I can't test LCD right now - too jet-lagged to hand-plug all the pins again right now :) but was going to make a clean ribbon one later once confirmed working.  If you have an extra handy, it would save me the trouble.

While I enjoy stretching the limits of what HW can do, and playing with UBL on the Printrboard was fun, UBL just too much of a bloated Windows Vista monstrosity to work on these Windows XP era boards so I think I'll put the LCD back on, run BILINEAR and call this one done.

I still need to finalize a PR for the pins_PRINTRBOARD_REVF.h (and possibly non-RevF if you might be able to test/or review it) to support the the MKS MINI12864, so I'll make reviewing all info in your wiki entry part of that when I do.  I was going to submit a page there myself but the RepRapWiki site won't take contributors unless you're already an insider it seems.  The contact info I finally found burried somewhere claimed to contact one of a handlful of admins but there was no way to do that either, so gave up.  Guess you're an insider ;)

fiveangle, 2017-06-24:
also, what do you guys think about my idea for

#define USE_INTERNAL_SD

To enable SD on external panel (disabling onboard SD for boards that have one) ? (or it's contrapositive)

It could be pushed into Configuration_adv.h if you don't want to clutter Configuration.h any more than it already is.

<details><summary>pins_PRINTRBOARD_REVF.h MINIPANEL changes</summary>
```
#if ENABLED(MINIPANEL)
  #if ENABLED(USE_INTERNAL_SD)
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN# 
    #define SDSS               20  //        10               B0  
    #define SD_DETECT_PIN      -1  // no auto-detect SD insertion on built-in Printrboard SD reader
  #else
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN#  Printrboard RevF Conn.   MKSLCD12864 PIN#  
    #define SDSS               11  //        36               C1                EXP2-13             EXP2-07
    #define SD_DETECT_PIN       9  //        34               E1                EXP2-11             EXP2-04
  #endif
  
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN#  Printrboard RevF Conn.   MKSLCD12864 PIN#  
    #define DOGLCD_A0           4  //        29               D4                EXP2-05             EXP1-04      
    #define DOGLCD_CS           5  //        30               D5                EXP2-06             EXP1-05
    #define BTN_ENC             6  //        31               D6                EXP2-07             EXP1-09
    #define BEEPER_PIN          7  //        32               D7                EXP2-08             EXP1-10
    #define KILL_PIN            8  //        33               E0                EXP2-10             EXP2-03
    #define BTN_EN1            10  //        35               C0                EXP2-12             EXP2-06
    #define BTN_EN2            12  //        37               C2                EXP2-14             EXP2-08
    #define LCD_BACKLIGHT_PIN  43  //        56               F5                EXP1-12       [unknown yet]
  //#define SCK                21  //        11               B1                ICSP-04             EXP2-09
  //#define MOSI               22  //        12               B2                ICSP-03             EXP2-05
  //#define MISO               23  //        13               B3                ICSP-06             EXP2-05            
  
  // encoder connections present
  #define BLEN_A 0
  #define BLEN_B 1
  #define BLEN_C 2

  // encoder rotation values
  #define encrot0 0
  #define encrot1 2
  #define encrot2 3
  #define encrot3 1

  // increase delays to max
  #define ST7920_DELAY_1 DELAY_5_NOP
  #define ST7920_DELAY_2 DELAY_5_NOP
  #define ST7920_DELAY_3 DELAY_5_NOP
#endif
```</details>

Bob-the-Kuhn, 2017-06-24:
To be able to edit/create RepRap WIKI pages I did the following:
1) got a RepRap forum account
2) stumbled on the create WIKI account page
3) copied the link to one of the admins
4) clicked on Private Messages at the top right of the page
5) pasted the link in & stated why I wanted a WIKI account
6) the next day they set one up for me.

----

Send your address to bob.kuhn@att.net

Here's the pinout of the cable.
[temp.zip](https://github.com/MarlinFirmware/Marlin/files/1099964/temp.zip)

commit ed90f755 (2017-05-03): Fix #6550