MarlinFirmware/Marlin #6555: Feature Request: Dynamic Hotend Cooling Fan Speed

timeshell, 2017-05-03:
I imagine I might get some flack for this but here it goes.

I have to slow down the cooling fan on my E3D v6 hotend in order to get it up to higher temperatures like 270.  Perhaps a feature could be to automatically adjust the heatsink cooling fan if the temperature won't rise during heat up.  Slow it down bit by bit until the heater block gets to temperature.

Thoughts on this?

psavva, 2017-05-03:
You should control your fan view your Slicer.
Which slicing software are you using?

oysteinkrog, 2017-05-03:
I think this is not a good idea, it's a hardware problem which should not be fixed in software.
Wrap/insulate your heater block, e.g. use a E3D sock.

timeshell, 2017-05-03:
Already use 24V heater cartridge AND E3D sock.

alexborro, 2017-05-03:
Use a 12V cartridge powered by 24V. Limit the output PWM to about 128 (50%)
and you will have about twice the cartridge rated power.

You will need to adjust the PID as well.
And make sure the cables can carry twice the cartridge rated current.

Cheers

Alex.

nealpalmer, 2017-09-26:
I asked for the same thing.  It should be pretty simple to have the cooling fan speed limited (i.e. PID algorithm) to keep the hot-end PWM at about 75%.

thinkyhead, 2018-02-09:
One wonders how this would affect PID tuning without first tuning with this feature enabled as part of `M303`.

boelle, 2019-02-18:
i wonder if the fan is not cooling the block instead of under the block

@timeshell have you check if the duct is designed correctly?

InsanityAutomation, 2019-02-18:
Adaptive fan slowing which was merged recently fulfills this request.

boelle, 2019-03-12:
@thinkyhead i think we can close this one

as @InsanityAutomation said its fullfilled by adaptive fan slowing

closed 2019-03-20 by timeshell