MarlinFirmware/Marlin #6560: Motors movements testing

vMeph, 2017-05-03:
Hi 
Im attempting to build my frirst 3d printer, at this point all i have is ramps.1.4,arduino mega and nema 17 motors, i was trying to test movements of motors by sending gcode files into it, but sense i dont have extruder,heatbed and sensors of temperature, when i send the files motors start working but then stops with alarm of min temp halted kill, so im guessing the alarem comes cause i dont have temp sensors, so i was looking into config files and i cant see how to disable the bed, extruder and temp sensors,i would say that most people when  star building a printer most like first things we want to do is test movments, would be nice to have a #define that alows to disable all heat,temp related things for testing motor movements purpose

Could you point me on the right direction.
Thanks

teemuatlut, 2017-05-03:
Set TEMP_SENSOR_0 and TEMP_SENSOR_BED to either 998 or 999.

vMeph, 2017-05-03:
@teemuatlut thanks alot that solved problem :)
now i have test motors all have work except the E0 motor extruder i got no idea why not working i have swaped drives and isnt a problem with the drivers or the motors no idea where to look when i try move extruder from LCD or software dont work, i notice that if i try to move by hand the shaft only the  extruder motor is lose
any ideas?

lrpirlet, 2017-05-04:
@vMeph 
I bet you are trying to run the extruder motor, not taking into account that the it is prevented to run before the filament is hot enough...

M302 should allow you to cold extrude http://reprap.org/wiki/RepRap_Firmware_G-Codes ...

Marlin implementation of M302: http://marlinfw.org/docs/gcode/M302.html

so, I would try M302 S0

enjoy...

vMeph, 2017-05-04:
@lrpirlet that nakes it work, great thanks alot mate :)

closed 2017-05-25 by vMeph