MarlinFirmware/Marlin #6562: Freeze on startup

asterycs, 2017-05-03:
Hello folks,

I wanted to try out the updated linear advance feature and thus updated to the `6283913` on RCBugFix. I updated my configs accordingly. When booting the new firmware my printer sort of freezes after reaching the status page. The hot end fan also starts running, but the lcd is totally unresponsive. Any ideas? The same thing happens when using example config files. The printer in question is a delta. Is there some new feature that might be causing this?

Tannoo, 2017-05-03:
Wrong MB selected? Or wrong LCD?

thinkyhead, 2017-05-04:
Start with a download of the current RCBugFix (https://github.com/MarlinFirmware/Marlin/commit/11b6a3cbb665b2dc5433ff0a56358289b5251ac7) and see if it fares any better. Some bugs have been patched today.

asterycs, 2017-05-04:
False alarm this time, it was indeed a configuration error. Thank you for the help!
Nice work on the new release by the way!

closed 2017-05-04 by asterycs