MarlinFirmware/Marlin #6563: Comparing int to float for temperature measurements

simon-jouet, 2017-05-03:
Hi,

So like I said in the previous post, I've been porting the code to ESP32 and was getting crashes in the temperature ISR. After quite a bit of debugging I finally figured out the issue was because of invalid comparisons.

So if you have a look at https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/temperature.cpp#L67 target_temperature and the like are defined as int, however in the isr while performing the comparisons, https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/temperature.cpp#L1965 the int value is compares with 0.0f.

Looks like this is dealt with by the Arduino compiler but might be causing some problems

Roxy-3D, 2017-05-03:
I don't think it is the comparison causing the problem.   The compiler will convert the types (with possible complaints about loss of precision).   I think it is more likely that we have an order of evaluation issue.   I guess we need to looks at the disassembled code and see what is being generated.   But looking at:
```cpp
      const int tdir = temp_dir[e], rawtemp = current_temperature_raw[e] * tdir;
      if (rawtemp > maxttemp_raw[e] * tdir && target_temperature[e] > 0.0f) max_temp_error(e);
```
What happens if tdir is not assigned until after rawtemp is assigned?   Are you able to look at the various values and see what values all the variables have on that if() statement?

For starters...  I might try:
```cpp
      const int tdir , rawtemp;
      tdir = temp_dir[e];
      rawtemp = current_temperature_raw[e] * tdir;

      if ( (rawtemp > (maxttemp_raw[e] * tdir)) && (target_temperature[e]>0 ) )
        max_temp_error(e);
```

simon-jouet, 2017-05-03:
Hi @Roxy-3D,

I actually did what you suggested before posting when I was debugging and it didn't solve the issue. Just to make sure I just tried again right now and the problem remains, when I use the proper types the issue disappear.

When I have slightly more time I might have a look at the asm listing, but my current *guess* right now is that the ESP compiler does some optimisations for the floating point unit that causes the issue. 

Anyway just wanted to report the issue as comparing int and float as it might cause someone else troubles and comparing int and float is never a really good practice. Feel free to close the issue if you think it shouldn't be addressed :)

Cheers,

Roxy-3D, 2017-05-03:
> I just tried again right now and the problem remains, when I use the proper types the issue disappears.

OK!   We need to change this in the code.   Other people messing with the 32-Bit stuff have been having problems and I wouldn't be surprised if this was at the root of it!

With the problem patched so it doesn't blow things up...   It might be good to add a new M command so you can cause the problem not at interrupt time.   What I'm thinking is make a new M command that declares and assigns values to simple variables of the types you see there...  and does the if() with the floating point comparison.     Having a small simple piece of code to duplicate the problem would be further proof you really found AND fixed the problem.    

But what I'm really thinking is...  we need to run  your .elf file through your processor's obj-dump program with a -S to get a disassembly of the code  (maybe with and without the floating point comparison).   

We need to understand what the compiler is thinking.    

Can you make a Pull Request to fix those types?

thinkyhead, 2017-05-03:
> What happens if tdir is not assigned until after rawtemp is assigned?

`rawtemp` depends on `tdir` so the compiler assigns `tdir` first.

You are still promoting your misunderstanding of `const` instead of listening to my explanation.

Roxy-3D, 2017-05-03:
> You are still promoting your misunderstanding of const instead of listening to my explanation.   

We are talking about a compiler bug here.   Is it possible that means the compiler is not behaving as it is supposed to?    If the compiler was doing the 'Right Thing', he would not be having a problem comparing an int to a float, would he?

thinkyhead, 2017-05-03:
A compiler bug is the last thing we should expect. Not the first.

thinkyhead, 2017-05-03:
>  I actually simply tried to compare `target_temperature[e] > 0.0` with nothing else 

Not also that on 32-bit "`0.0`" will be a `double`, while `0.0f` will be a `float`. Try these alternatives and see what happens…

thinkyhead, 2017-05-03:
>  I actually simply tried to compare `target_temperature[e] > 0.0` with nothing else 

Not also that on 32-bit "`0.0`" will be a `double`, while `0.0f` will be a `float`.

thinkyhead, 2017-05-03:
Seems most temperatures are integers in this context. I'll make a quick scrub-through to make sure there are only integer assignments and comparisons, for safety, and post a PR shortly.

Roxy-3D, 2017-05-03:
> I'll make a quick scrub-through to make sure there are only integer assignments and comparisons, for safety, and post a PR shortly.

It would be nice if the changes went into both RCBugFix and 32-Bit-RCBugFix-New.

thinkyhead, 2017-05-03:
The 32-bit branch will need to be re-built manually based on the latest code. I will go through that process shortly after the release of 1.1 — which I expect will be tonight, barring the appearance of any new blocker issues.

thinkyhead, 2017-05-03:
> I actually simply tried to compare target_temperature[e] > 0.0 with nothing else and it made the processor crash.

Bad compiler! It probably just hasn't been attuned for code as phat and complex as Marlin.

bobc, 2017-05-03:
There is nothing wrong with comparing floats and integers, and I have never
heard of it being called bad practice. I'd be very surprised if the
compiler couldn't handle it.
The key thing here is that in an interrupt, "Tread carefully, bad things
can happen". I suspect it is more likely a stack corruption or such like.
There is probably no harm in using ints for temperatures, oversampling
might need more range, but it would be useful to get to the bottom of the
issue, if it could happen anytime anywhere.

Roxy-3D, 2017-05-03:
> " Floats use the FPU while doubles are calculated in software. For reasons, using the FPU inside an interrupt handler is currently not supported "

OK!  But I think I would argue to compiler should warn that the code has an unsupported construct (float inside ISR).

Thanks Bobc!!!

bobc, 2017-05-03:
This is one of those alligator infested swampy areas that lies outside "standard C" and in "implementation defined behavior". Interrupts are a very platform dependent thing. Unfortunately, when the alligators appear, often the tour guide is nowhere to be seen :)

I've been bitten by this sort of thing at work, eventually the response from the vendor was "oh, those useful math library functions can't be used inside interrupt!" Nice...

thinkyhead, 2017-05-03:
So, there you go, a nice easy solution. Just cast all the `float` variables and values used in the ISRs to `double` and you're good to go!

For the HAL we'll have to add `typedef`s for interrupt-safe decimal numbers (`isr_decimal`). They can be `float` on AVR, but `double` on platforms that have this FPU limitation.

Also, we'll need a `typedef` for the one that calculates faster — `fast_decimal`, I suppose.

Roxy-3D, 2017-05-03:
> For the HAL we'll have to add typedefs for interrupt-safe decimal numbers (isr_decimal). They can be float on AVR, but double on platforms that have this FPU limitation.

They can be double on AVR also.   double and float are the same size so it would make sense to pick a data type that works in both places.

closed 2017-11-01 by thinkyhead