MarlinFirmware/Marlin #6571: Reboot after print selection via LCD or OctoPi

cryptelli, 2017-05-04:
Hi Guys,

This isn't my first rodeo, as I've successfully flashed the firmware for my printer numerous times in the past starting with RC3 to RC6 and then RC8 to RCBugFix (1.0.1 branch) which is what I've been running for the last few weeks.

After seeing that Marlin 1.1.0 was released I decided to upgrade as I figured it would no doubt have a bunch of stuff added since I updated and after migrating my configuration over I uploaded it.

This is where things went downhill, after the first reboot from the upgrade and the second from the connection to OctoPi I selected a print, and was surprised to see the bot in a reboot loop. In an effort to troubleshoot the problem I decided to try printing from the SD card alone, same problem...

Went back into my configuration and checked everything was present and correct, realised I had a double entry for the hotend PID tuning, after commenting out the default for my own calibration i'd previously run I re-uploaded and checked, same problem.

Problem is non existent if I flash the printer back to my patched RCBugFix.

Advise?

VanessaE, 2017-05-04:
I'm seeing the same problem here.

NiklasOlofsson, 2017-05-04:
Also having a "boot loop" with the latest RCBugFix when I am connected using Octoprint. Not connected, it stays on but as soon as I connect to the printer, the boot loop starts.

bgort, 2017-05-04:
I'm seeing this problem also -- same here, when connected via serial (octoprint) it reboots.  Will do some digging around shortly.

cryptelli, 2017-05-04:
Didn't think I was mad! Thanks for adding your experiences people. Hopefully those whom know the workings better than I do will be able to get to the bottom of this issue.

bgort, 2017-05-04:
241bdffe65ea4ef8c776bc68b75a77c07cb7e3bf works for me, while 2658cc707ad03600c364cafb7adbfa6261c3f577 does not.  So our problem is in 2658cc707ad03600c364cafb7adbfa6261c3f577 or 1b2c7ec20a23af57da50dafa672283ff09bbaf51.

Unfortunately nothing stands out yet, but still looking.

Neverther, 2017-05-04:
Same issue, worked on RCBUGFIX
Connecting via octoprint on bufix-1.1.x r8299:

Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x7323c690, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CoreXY EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: ok
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M117 IP 192.168.0.101
Recv: ok
Send: M105
Recv: ok T:start
Send: M105
Recv: echo:Marlin bugfix-1.1.x
Recv: 
Recv: echo: Last Updated: 2017-05-04 12:00 | Author: Never, CoreXY
Recv: Compiled: May  4 2017
Recv: echo: Free Memory: 3538  PlannerBufferBytes: 1232
Recv: echo:V37 stored settings retrieved (453 bytes)
Recv: echo:  G21 ; Units in mm
Recv: 
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z2560.00 E500.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z2.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y3000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H180 B70 F0
Recv:   M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z0.00
Recv: echo:SD init fail
Recv: ok T:start
Send: M105
Recv: echo:Marlin bugfix-1.1.x
Recv: 
Recv: echo: Last Updated: 2017-05-04 12:00 | Author: Never, CoreXY
Recv: Compiled: May  4 2017
Recv: echo: Free Memory: 3538  PlannerBufferBytes: 1232
Recv: echo:V37 stored settings retrieved (453 bytes)
Recv: echo:  G21 ; Units in mm
Recv: 
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
[... 153 lines cut ...]

-//#define REVERSE_MENU_DIRECTION
+#define REVERSE_MENU_DIRECTION

-//#define INDIVIDUAL_AXIS_HOMING_MENU
+#define INDIVIDUAL_AXIS_HOMING_MENU

-//#define SPEAKER
+#define SPEAKER

-//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100
+#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 50

-//#define REPRAP_DISCOUNT_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_SMART_CONTROLLER

Configuration_adv.h:
-//#define BABYSTEPPING
+#define BABYSTEPPING

-//#define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
+#define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.

-//#define FILAMENT_CHANGE_FEATURE
+#define FILAMENT_CHANGE_FEATURE

colinrgodsey, 2017-05-04:
Same issue. Worked with RCBUGFIX, not with 1.1.x. Not getting anything useful out of the serial logs...

MKS Base 1.4
Endstop interrupts
LIN_ADVANCE
BLTouch

colinrgodsey, 2017-05-04:
Might be a long shot, but looking over the differences in the changes mentioned above:

https://github.com/MarlinFirmware/Marlin/compare/241bdffe65ea4ef8c776bc68b75a77c07cb7e3bf...2658cc707ad03600c364cafb7adbfa6261c3f577

There doesnt seem to be anything related to SD or whatnot, but there does seem be a shift from using floating point to ints in a few places. I know normally in C you'll get a NaN for divide by zero for floating point, but you'll get a segfault if you do an integer divide by 0.

Not that familiar with Arduino/Atmega... but it sounds like it could be a possibility. Going to fiddle around and see...

bgort, 2017-05-04:
Yeah..nothing really stands out at all, but that could be it.  Don't think this has anything to do with SD, as I don't even have SD support enabled.

I did find a few ampersands where they didn't belong (in the arguments to setTargetHotend and setTargetBed, particularly), and a few other little things, but fixing them didn't resolve the issue.

Have a few calls, an errand, etc., so am going to have to push this for a bit, unfortunately.

bgort, 2017-05-04:
Saw a fair bit of implicit casting that might possibly be causing a problem, too, but haven't yet looked into that.  Maybe something to consider as you fiddle.

colinrgodsey, 2017-05-04:
@bgort cool, ill keep an eye out

BillyBob54, 2017-05-04:
This, or at least a related issue, may go further back, see #6128; I have not had a chance to load up 1.1.0 and try.

colinrgodsey, 2017-05-04:
Looks like m105 is causing the crash:

```
Send: M105
Recv: ok T:start
```

You get "T:" then immediately just get the 'start' from the reboot.

bgort, 2017-05-04:
Mine isn't getting that far:

```
Send: N0 M110 N0*125
Recv: start
```

Grogyan, 2017-05-04:
This may be due to the on board 5V regulator, or polyfuse, going into
thermal shutdown.
If you have a DMM, you can measure the 5V rail to see if it does drop out

colinrgodsey, 2017-05-04:
found it! Ultimately I think the SERIAL_PROTOCOL_F function was not handling the new uint16_t values correctly. I switched some of the inline methods in temperatures.h back to float/int and things work fine. Going to clean things up a bit and narrow down the changes to just the core.

bgort, 2017-05-04:
That's funny!  Just realized that myself and compiling to test an int instead of uint16_t in the output....

Roxy-3D, 2017-05-04:
There was a similar problem when char's where turned into int8_t types.   The SERIAL_ECHO() macro passed them to myserial.serial() and the wrong thing happened.   If I remember right, instead of printing characters, numbers came out.

Hertog99, 2017-05-04:
@colinrgodsey's changes fixed the rebooting issue for me. Don't know if it has possible side-effects?

colinrgodsey, 2017-05-04:
@Hertog99 well, technically can't be sure of the quality of the original build, couldnt run it ;) so not sure. buyer beware!

Roxy-3D, 2017-05-04:
> found it! Ultimately I think the SERIAL_PROTOCOL_F function was not handling the new uint16_t values correctly. I switched some of the inline methods in temperatures.h back to float/int and things work fine. Going to clean things up a bit and narrow down the changes to just the core.

The changes from float to int were made to help the migration to 32-bit processors.  There was a problem found a few days ago where float operations done at interrupt time caused the processor to crash.   double operations were allowed because they are done in software.  But the float operations happen inside the floating point unit and nothing can touch that while servicing an interrupt.

It probably would be best to not switch back to float.   Perhaps changing from SERIAL_PROTOCOL_F() to just SERIAL_PROTOCOL() would make sense?

Hertog99, 2017-05-04:
Just tried a print, bed won't heath up when sending print via octoprint, also, I heard my BLTouch click, guess it did a reboot. (Jup, just checked, sending a print via Octoprint/Octopi causes the printer to reboot) Some more digging is needed.

bgort, 2017-05-04:
Suspect we need to fix the actual underlying issue, likely in MarlinSerial::printNumber()

Roxy-3D, 2017-05-04:
> Suspect we need to fix the actual underlying issue, likely in MarlinSerial::printNumber()

Yeah...  But for sure...  It isn't valid to be calling as SERIAL_PROTOCOL_F((int),(int)).  Won't changing to a SERIAL_PROTOCOL() be just fine?   It will print an integer just fine....

bgort, 2017-05-04:
Oh -- you're absolutely right.  Trying that now.

colinrgodsey, 2017-05-04:
@Roxy-3D wow, thats brutal about the interrupt space stuff lol

ill revert my stuff and just try switching some of the APIs over to SERIAL_PROTOCOL. See if I can find any other that explode

colinrgodsey, 2017-05-04:
Is there some issue with the method overloads on MYSERIAL.print? Floats work fine, even tho theres only an overload for double, so not sure why uint_16t wouldnt be implicitly upcast to int. 

This just some compiler oddity?

bgort, 2017-05-04:
No, it's as simple as SERIAL_PROTOCOL_F being for floats, while SERIAL_PROTOCOL is for everything else.  I totally missed that when I looked at the macro and associated function.

Roxy-3D, 2017-05-04:
>  even tho there's only an overload for double, so not sure why uint_16t wouldnt be implicitly upcast to int.

Well...   I bet we are missing the overloaded function definition for uint_16t.   (But I haven't looked.)

bgort, 2017-05-04:
We are, but I tried to add it, but the compiler wouldn't allow it.  uint16_t = int, so it's not a valid overload.

bgort, 2017-05-04:
Actually, I think what was happening is that because the first argument wasn't a float, it was calling print(int, 1) .. where 1 is the base of the number....  And obviously that isn't going to work.

Roxy-3D, 2017-05-04:
> uint16_t = int

Well...   That doesn't make sense.    The two are not the same.    There must be something we are missing???    You must be able to have a declaration for 
```cpp
void func(int);
and
void func(unsigned);
```
That would only make sense?????

bgort, 2017-05-04:
Sorry - I meant int16_t = int.  I blame my fingers...

colinrgodsey, 2017-05-04:
Getting a bit off topic here but... doesn't it kind of suck that we're not getting fractional temperatures now?

```C
    static int16_t degHotend(uint8_t e) {
      #if HOTENDS == 1
        UNUSED(e);
      #endif
      return current_temperature[HOTEND_INDEX];
    }
```

current_temperature is a still a float, why are we removing the fractions? does this just effect the serial API?

houseofbugs, 2017-05-04:
I tried the fix in here and that did not change anything. I am having this issue on a Melzi and MKS boards.

Replicated using SD and USB. Both caused a reset.

bgort, 2017-05-04:
@houseofbugs:  Did you pull down the latest bugfix-1.1.x?  ae7c602

VanessaE, 2017-05-05:
I haven't tried @Roxy-3D's changes yet, but if that works, I would highly recommend issuing a new "hotfix" release tagged 1.1.1 (or 1.1.0.1?) ASAP.

cryptelli, 2017-05-05:
I can also confirm that the patch works, nice work team and I agree with @VanessaE, this seems to be a widespread issue which shouldn't be present in 1.1.0.

Also, the temperature graph isn't as granular like @colinrgodsey pointed out, any idea if that is going to be how it is moving forward or can we can back to having fractional numbers?

closed 2017-11-01 by thinkyhead