MarlinFirmware/Marlin #6573: Guidance on new branches

bgort, 2017-05-04:
Congratulations on the release of 1.1.0!  👍 

Now that the release is finished, I see there are 2 new branches - '1.1.x' and 'bugfix-1.1.x'.  Is there guidance on what these branches will be used for?  Which should we do PRs against?  Are we separating new features from bugfixes, with new features and bugfixes applied to '1.1.x' and bugfixes only to 'bugfix-1.1.x' ?

Thanks!

Tannoo, 2017-05-04:
According to the new Readme:
```
To submit patches and new features for Marlin 1.1 check out the bugfix-1.1.x branch, add your commits, and submit a Pull Request back to the bugfix-1.1.x branch. Periodically that branch will form the basis for the next minor release.
```

bgort, 2017-05-04:
Ah, thanks!  Hadn't yet seen the readme.

closed 2017-05-04 by bgort