MarlinFirmware/Marlin #6575: SD card printing

longle2710, 2017-05-04:
When i click on a file to print, the machine rebooted as of i hit the reset button

NiklasOlofsson, 2017-05-04:
It might be the same problem as described here:
https://github.com/MarlinFirmware/Marlin/issues/6571

longle2710, 2017-05-04:
Thank you, I hope we can solve it soon.

houseofbugs, 2017-05-04:
I have this same issue on my MKS and Melzi boards.

stevereno30, 2017-05-05:
I have the exact same problem with my machine. I'm using an Arduino Mega with a RAMPS 1.4. 

The odd thing is I did not have this problem at all with 1.1.0 RC8.

fred2088, 2017-05-05:
same here, got reboot from 1.1 release, no problem revert to rc8

bgort, 2017-05-05:
Have you guys tried the bugfix-1.1.x branch?

stevereno30, 2017-05-05:
I just loaded the bugfix-1.1.x branch and it works fine. Thanks for the tip!

closed 2017-11-01 by thinkyhead