MarlinFirmware/Marlin #6576: M206 has no effect

SuperMechaCow, 2017-05-04:
Suddenly, without warning, the M206 home offset stopped working and nothing I do has any effect on it.

The firmware is set to offset X and Y homing positions my -23 and -30 respectively. My toolhead sits outside of the print area. When I home and then tell it to move the toolhead it normally moves it to the bottom-left of the bed and sets that position to 0,0 before completing the move, as expected.

However now, with no warning or changes, it always uses 0 and 0 as the X and Y home offsets no matter what I do. This causes the inductive Z-probe to miss the bed by 30 units on the Y axis and causes a bed crash while probing with G29. It also causes the bed to overshoot and grind when moving the Y axis all the way forward.

M502 restores the M206 offset to 0,0,0, which is not what is set in the firmware. Changing M206 can be seen in M503 and in the EEPROM saved settings, but it has zero effect on the actual home offset. The Tool is always homing to the wrong position no matter what.

You can see in this image a vague white line to the left of the toolhead in the painter's tape. This is exactly 10 units from 0,0. The image is where the toolhead keeps homing.

I am controlling this through Octoprint remotely, but again, I have changed nothing in the last 24 hours that would effect the home offset.

FATALHERMIT, 2017-05-10:
@SuperMechaCow what image?

ellisgl, 2017-09-21:
I'm having this same issue with 1.1.5

thinkyhead, 2017-11-27:
Please test with current `bugfix-1.1.x` and if the issue is fixed please close.

boelle, 2019-02-18:
@SuperMechaCow is this still an issue with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
@SuperMechaCow i just used M206 and it works fine

SuperMechaCow, 2019-04-19:
Please close. I fixed this so long ago I forgot the solution

p3p, 2019-04-19:
That would be your job @SuperMechaCow, you will see the close button at the bottom.

SuperMechaCow, 2019-04-19:
I am such an _idiot_ today

closed 2019-04-19 by SuperMechaCow