MarlinFirmware/Marlin #6579: Misleading SanityCheck for Z probe endstop

colinrgodsey, 2017-05-04:
This sanity check was driving me nuts. Kept doing what it was asking, but kept getting the error. Finally gave it a good long stare....

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/SanityCheck.h#L514

```C
    #elif ENABLED(Z_MIN_PROBE_ENDSTOP_INVERTING) != ENABLED(Z_MIN_ENDSTOP_INVERTING)
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
    #endif
```

It says it's required that they match, but the test itself is guaranteeing that only one is set!

etsmith3, 2017-05-06:
I'm having this same issue, both variables are set to true (checked and double checked).

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define **Z_MIN_ENDSTOP_INVERTING true** // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define **Z_MIN_PROBE_ENDSTOP_INVERTING true** // set to true to invert the logic of the probe.

SanityCheck.h:517: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."

       #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."

etsmith3, 2017-05-06:
I commented my sanity check to look like this for now
```/**
   * Require pin options and pins to be defined
   */
  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #if ENABLED(Z_MIN_PROBE_ENDSTOP)
      #error "Enable only one option: Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN."
    #elif DISABLED(USE_ZMIN_PLUG)
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires USE_ZMIN_PLUG to be enabled."
    #elif !HAS_Z_MIN
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
    /* This was set as a temp fix for below not working
     *  https://github.com/MarlinFirmware/Marlin/issues/6579
    #elif ENABLED(Z_MIN_PROBE_ENDSTOP_INVERTING) != ENABLED(Z_MIN_ENDSTOP_INVERTING)
      #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."
    */
    #endif    
  #elif ENABLED(Z_MIN_PROBE_ENDSTOP)
    #if !HAS_Z_MIN_PROBE_PIN
      #error "Z_MIN_PROBE_ENDSTOP requires the Z_MIN_PROBE_PIN to be defined."
    #endif
  #elif DISABLED(PROBE_MANUALLY)
    #error "You must enable either Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use a probe."
  #endif```

colinrgodsey, 2017-05-08:
At least for me, I just had to set one of: Z_MIN_PROBE_ENDSTOP_INVERTING or Z_MIN_ENDSTOP_INVERTING. Im not totally sure what the difference is, but that check definitely suggests it wants only one. Carriage hasn't crashed into the bed yet, so guess it works lol.

Does anyone have any insight on what the expected functionality is?

David-en8, 2017-06-26:
If you are using a capacitance probe then this is used for the Z endstop reading and for bed leveling.
The digital signal our of the probe should be identical for both logically as it is the same device.
In my case I converted my probe which was a 6-36 volt version to use an opto isolator, this inverted the original logic but gave me a little more sensitivity.

thinkyhead, 2018-02-07:
Seems to work fine:
```cpp
  #define TEST_ME_1 true
  #define TEST_ME_2 true
  #if ENABLED(TEST_ME_1) != ENABLED(TEST_ME_2)
    #error "TEST_ME_1 is not equal to TEST_ME_2!"
  #endif
```

closed 2018-02-07 by thinkyhead