MarlinFirmware/Marlin #6582: Marlin 1.1.x - E3D PT100 and Amp

TGMods, 2017-05-04:
I have just upgraded to Marlin 1.1.x and I have a problem with the temperature reported on my LCD as seen in the second picture. The temperature reported is 100C and I use Temperature Sensor 20
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 11

I reinstalled RCBugfix dated 23/03/2017 and the temperature reported is 28C as seen in the first picture. I use the same temperature sensor number.
#define TEMP_SENSOR_0 20
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 11

Has there been a change to the temperature sensor numbering in Marlin 1.1.x?

[image]

[image]

Bob-the-Kuhn, 2017-05-04:
Check to see if the internal pullup is enabled for that pin.  That'll throw the reading off in the positive direction.

TGMods, 2017-05-05:
@Bob-the-Kuhn Sorry, my bad. I forgot to change the pins.h file.
Thank you anyway

closed 2017-05-05 by TGMods