MarlinFirmware/Marlin #6583: What is the difference of MESH_BED_LEVELING and MANUAL_BED_LEVELING

batata004, 2017-05-04:
Hi,

I realized that in order to me be able to use MANUAL_BED_LEVEING I also have to enable MESH_BED_LEVELING in the firmware. What is the diffenrece?

This page -> https://github.com/MarlinFirmware/Marlin/wiki/Mesh-Bed-Leveling   is very clear but I cant understand the difference of both options.

The MANUAL_BED_LEVELING, to me, is very easy to undersstand: it probes some points in the print area and I am requested to raise/lower the nozzle using a paper to check the distance. But what is MESH_BED_LEVELING for?

Tannoo, 2017-05-04:
From the wiki:
`MESH_BED_LEVELING` will enable mesh bed leveling.
`MANUAL_BED_LEVELING` will add the menu option for bed leveling.

From `configuration.h`:
```
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
```
```
/**
 * Use the LCD controller for bed leveling
 * Requires MESH_BED_LEVELING or PROBE_MANUALLY
 */
//#define LCD_BED_LEVELING
```
It looks like `MANUAL_BED_LEVELING` was changed to `LCD_BED_LEVELING`.

batata004, 2017-05-04:
Ok, I just dont get what this says:

`MESH_BED_LEVELING` will enable mesh bed leveling.

Ok, will enable mesh bed leveling. So what? I can only use it if I enable LCD_BED_LEVELING right?

Tannoo, 2017-05-04:
Enableing both will give the option to start the leveling via the LCD.
If `LCD_BED_LEVELING` is not enabled, but `

batata004, 2017-05-04:
Great man! Just one question: I have the inductive sensor to use with AUTO BED LEVEING and it works fine. BUT Sometimes I dont want to use the inductive sensor (beucase I am using glass in the bed) so I would like to use th MANUAL BED LEVELING.

HOWEVER WHEN I TRIED enabling MESH_BED_LEVELING it does not allow me to upload the firmware cause it already says I am using auto bed leveling! Cant I have both options so I dont need to flash a new firmware everytime I am using glass/metal in the bed?

epatel, 2017-05-04:
Actually, MESH_BED_LEVELING was possible without the LCD. I am not sure about the current status. One use a few G29 codes, like S1 to start a probing sequence, use G1 to lower/raise and then S2 to go to next point and repeat until done.

batata004, 2017-05-04:
Good to know it! Do you know if using BILINEAR AUTO LEVELING already makes use of MESH?

Tannoo, 2017-05-05:
Bilinear and mesh cannot be enabled at the same time.

Tannoo, 2017-05-05:
```
/**
 * Choose one of the options below to enable G29 Bed Leveling. The parameters
 * and behavior of G29 will change depending on your selection.
 *
 *  If using a Probe for Z Homing, enable Z_SAFE_HOMING also!
 *
 * - AUTO_BED_LEVELING_3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 *
 * - AUTO_BED_LEVELING_UBL (Unified Bed Leveling)
 *   A comprehensive bed leveling system combining the features and benefits
 *   of other systems. UBL also includes integrated Mesh Generation, Mesh
 *   Validation and Mesh Editing systems. Currently, UBL is only checked out
 *   for Cartesian Printers. That said, it was primarily designed to correct
 *   poor quality Delta Printers. If you feel adventurous and have a Delta,
 *   please post an issue if something doesn't work correctly. Initially,
 *   you will need to set a reduced bed size so you have a rectangular area
 *   to test on.
 *
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

batata004, 2017-05-05:
Why they cant be enabled at the same time? One is designed to manual leveling and another to auto leveling... why cant the user has both options, sometimes the user wants to manually level the bed (for reasons I pointed earlier) and sometimes he wants the practicity of auto leveling...

Tannoo, 2017-05-05:
Nope.. It can't happen. You have to re-flash to change them.

That's the way it is and I don't think that's going to change because it would take up too much memory, not to mention conflicts.

closed 2017-05-06 by batata004