MarlinFirmware/Marlin #6585: Marlin Updates with out config?

Radiu, 2017-05-04:
hi all
i would like to know how you guys do it every time you download RCBugFix updates, do you have to re-config machine every time or do you use other way to update with out reconfigure?

thanks

Tannoo, 2017-05-04:
We all use different programs, but if the differences are small, you can just copy the old config files into the new version. If going from 1.0.x to 1.1.x, you are best with a visual side-by-side comparison.

I use a program call WinMerge to look at the differences between files.

Radiu, 2017-05-04:
@Tannoo just tried now WinMerge sounds like a cool program 👍 
so im guessing now with the new release Marlin-bugfix-1.1.x will be the new place for the new updates fix
do you know if  to update from RCBugFix to Marlin-bugfix-1.1.x  you need to clean EEPROM and reinstall all new?

Tannoo, 2017-05-04:
Get `configuration.h` and `configuration_adv.h` up to date first.
Flash the firmware.
Send M502, M500 and M501, in that order.
You are good to go then.

_best_ gcode sender?
Well, there is no best, but the popular ones are Pronterface, Repetier, and Octoprint. There are others, one that comes to mind is MatterControl.

Radiu, 2017-05-04:
@Tannoo  when i issue M500 says on echo error:EEPROM disable in red, is that normal?

if possible and you could explain in short words what the sequence Send M502, M500 and M501 is actually doing i would appreciate 

thanks alot for the help you been doing, im new on this, i come from CNC grbl content, but in 3D printers there is so many information on it and M commands

Tannoo, 2017-05-04:
```
// The microcontroller can store settings in the EEPROM, e.g. max velocity...
// M500 - stores parameters in EEPROM
// M501 - reads parameters from EEPROM (if you need reset them after you changed them temporarily).
// M502 - reverts to the default "factory settings".  You still need to store them in EEPROM afterwards if you want to.
//define this to enable EEPROM support
//#define EEPROM_SETTINGS
```

closed 2017-05-14 by Radiu