MarlinFirmware/Marlin #6589: Z axis gets inverted after first layer

discojon, 2017-05-05:
Lulzbot Taz 5

After the first layer gets printed, all subsequent layers are inverted driving the nozzle into the print.  Prior to the print I can jog and home successfully and everything works properly.   Initiate a print, first layer or skirt (if enabled) prints fine.  At the first +z motion something is happening and inverting the Z axis.  If i have a skirt on the part, the skirt prints fine, then the actual part won't print as the nozzle is moving to -Z.  The display is still reading the proper Z height.  After aborting the print, the jog controls are still reversed UNTIL I Jog -Z.  Once the -Z jog happens the directions are fixed until I try to print again.

discojon, 2017-05-05:
https://pastebin.com/Vh6Wa6YD

discojon, 2017-05-05:
Known good Gcode on previous firmware 
https://pastebin.com/WK8wSqSH

FHeilmann, 2017-05-05:
did you do a M502 & M500 after flashing the newer firmware?
the g-code and configuration looks ok from what I can tell.

discojon, 2017-05-05:
I just tried loading and saving the Config with M502 and M500.  Same behavior.

discojon, 2017-05-05:
Youtube link to video of the issue

https://www.youtube.com/watch?v=MD41KBHc8w8

james94jeans2, 2017-05-05:
do you use z_fade_height with any sort of leveling?

discojon, 2017-05-05:
I didn't explicitly turn it on anywhere in the config.  I'm using ABL as can be seen in the video.

james94jeans2, 2017-05-05:
Okay, then it won't be an issue concerning the fade.

Bob-the-Kuhn, 2017-05-05:
Is this only a problem when the heaters are on?

You could also try removing the filament & do some cold extrudes to see if it happens then.

discojon, 2017-05-05:
It has nothing to do with the extruder.  You can see the extruder still trying to extrude but the nozzle is touching the bed preventing extrusion.  The z axis is moving down after the skirt gets printed.

thinkyhead, 2017-05-05:
Try lowering the max speed on your Z axis, and perhaps the Z acceleration, and see if it makes any difference.

Bob-the-Kuhn, 2017-05-05:
The cold extrusion request was to see if the problem only occurred when the E stepper was active.

discojon, 2017-05-05:
I'll do some experiments later this evening and report back.

knobik, 2017-05-06:
I have the same problem but with upper layers, it has to do something with z-hop. Sometimes it goes up, sometimes it goes down like in this video i made https://youtu.be/qh-158TpTEc

discojon, 2017-05-06:
I am using z hop also

knobik, 2017-05-06:
I'll attack gcode and config as soon as my print finishes.

discojon, 2017-05-06:
The gcode I posted is only the first couple layers. And it works fine on RC-2

james94jeans2, 2017-05-06:
When z hop causes the issue, maybe your steppers are skipping. Try reducing the z-max Speed.

knobik, 2017-05-06:
Nope, they are not skipping, if you would saw the video, you would notice that the hotend goes back into position perfectly after bad z-hop. In my opinion as a programmer there is a problem with the code after guys porting the variables for a easier 32 bit convertion. There is something fishy going on cuz before i updated to last RCBugFix, everything went good.

discojon, 2017-05-06:
I don't believe mine are skipping either.  The issue always occurs after the first zhop. The reported z height is correct. I'll try reducing acceleration and z speeds though

knobik, 2017-05-06:
@discojon are you using UBL?

discojon, 2017-05-06:
Nope. ABL.

Sebastianv650, 2017-05-07:
I just did a quick test without any z hop and with FW retract z hop, both works but I'm not using any kind of automatic bed leveling.
From my point of view, as both of you use some kind of leveling, the question is if the problem persists if you disable ABL/UBL what ever and, if leveling is enabled, does the problem go away when switching from slicer generated z hop to FW retract z hops.

Then we would know if the error is related only to bed leveling, or if it's something generic but then it should be also repeatable with FW retract.

knobik, 2017-05-07:
cant test it right now cuz my printer is in pieces (literaly), but i can say that i was using slic3rs z-hop, and not the firmware version.

discojon, 2017-05-08:
Additional testing shows that with all forms of ABL disabled, and Z-Hop turned off in the slicer the print works fine.

discojon, 2017-05-09:
And it's not working again with the attached config and gcode.  It seems as though the Z axis isn't increasing in height with each layer, although the lcd displays the proper height.  The nozzle is being driven in to the printed material.
[I7.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/988224/I7.gcode.zip)
[Marlin 1.1 Taz.zip](https://github.com/MarlinFirmware/Marlin/files/988230/Marlin.1.1.Taz.zip)

discojon, 2017-05-10:
#5486  

This seems to be the issue

discojon, 2017-05-10:
And this seems to have been patched yesterday in bugfix.  I'll check it this evening.

#6656

FHeilmann, 2017-05-11:
Be aware that my fix isn't merged yet

discojon, 2017-05-12:
I can confirm that turning off Force retraction between layers in S3D fixes the issue.

knobik, 2017-05-12:
@discojon did you try @FHeilmann commit ?

discojon, 2017-05-12:
@knobik no I don't know how to merge it on my own

knobik, 2017-05-12:
git clone marlin, then add new remote from @FHeilmann repository, then fetch the branch and merge it into bugfix

discojon, 2017-05-15:
Well I am using the latest rc bugfix that I thought had the z-hop fix merged but it's definately still there.  With z-hop off, prints complete fine, but if z-hop is on either in FW retract or gcode, it crashes  after the first hop

thinkyhead, 2017-05-17:
Get the current `bugfix-1.1.x` code. Try lowering your Z axis jerk, acceleration, and manual movement speed values and see if this has any positive effect on the issue. If the Z axis tries to move too fast, it may do strange things.

boelle, 2019-02-18:
@discojon have you tried the latest bugfix 2.0?

discojon, 2019-02-19:
I went back to the lulzbot branch and forked my own from there.  This can be closed

closed 2019-02-19 by discojon